DSheirer/sdrtrunk #1851: Request: Add tuner id column to Now Playing grid

louisik1, 2024-02-12:
I would appreciate having a column to tell me which tuner is pulling each channel, maybe somewhere near the Channel and Frequency columns
[image]

In my case, I'd like to see the same Id that is shown on the Tuners tab

[image]

and also shown on the 'Preferred Tuner' option
[image]

I have set my tuner id's (via rtl_eeprom.exe) to help me know which one I'm dealing with. The USB device location id seems unreliable.

Much thanks

louisik1, 2024-02-12:
Oh I forgot, but I also edited the names in device manager but these names don't seem to show up anywhere in sdrtrunk.  Could be nice if they were accessible there too.  Device Manager shows this name via the "Display Name" and "Friendly Name" properties.
[image]

I hadn't thought about it until now but what might be super awesome is a way for users to name their devices in sdrtrunk (tied to each devices distinct guid aka "Container Id"), and save the names in the config file.

Thanks again for reading.

DSheirer, 2026-11-16:
Added #314

closed 2026-11-16 by DSheirer