DSheirer/sdrtrunk #1860: LIBUSB_ERROR_BUSY - 5 million times with configuration using dual Airspy SDR radios, ultimately hangs sdr trunk

n5lbr, 2024-02-20:
**sdrtrunk Version**
v0.6.1-beta-1 on Windows 11

**Describe the bug**
Use SDR trunk to monitor / capture a P25 system and upload to broadcastify.  After a couple of days sdrtrunk will lose connection to broadcastify and the console will be filled with errors similar to:
2024-02-20 06:14:13.669 ERROR i.g.d.s.t.u.USBTunerController - Attempt to resubmit previous error USB transfer buffer failed with status [LIBUSB_ERROR_BUSY] - this may be temporary and has happened [5261469] times so far.  Transfer buffer holding queue (error/total) [4/8]  [1GB/1GB 63%]

**To Reproduce**
Steps to reproduce the behavior:
Start sdrdrunk
Select Auto-Start the channel to monitor
Let it run for a couple of days.
After about 12 hours, the errors above will be in the console.  
After 3 days, sdrtrunk will disconnect from broadcastify and have to be re-started.  

**Expected behavior**
No disconnection from uploading to broadcastify, resolve LIBUSB_ERROR_BUSY errors on its own.

**Screenshots**
Console log filled with
2024-02-20 06:14:13.669 ERROR i.g.d.s.t.u.USBTunerController - Attempt to resubmit previous error USB transfer buffer failed with status [LIBUSB_ERROR_BUSY] - this may be temporary and has happened [5261469] times so far.  Transfer buffer holding queue (error/total) [4/8]  [1GB/1GB 63%]

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: Windows 11 pro 64 bit, 22631.3155
 - CPU Cores: [e.g. 4 cores] i7, 3.4Ghz
 - RAM:[e.g. 16GB] 64GB
 - Two USB AirspyR2 dongles connected to the system, being used to monitor the same P25 system.

**Additional context**
Add any other context about the problem here.

n5lbr, 2024-02-21:
I restarted sdrtrunk after submitting this and ~24 hours later, I see:

2024-02-21 06:09:53.842 ERROR i.g.d.s.t.u.USBTunerController - Attempt to resubmit previous error USB transfer buffer failed with status [LIBUSB_ERROR_BUSY] - this may be temporary and has happened [13129780] times so far.  Transfer buffer holding queue (error/total) [4/8]  [1GB/1GB 65%]

Disabling one of the tuners (second in the list), stops the error reporting / scrolling.

n5lbr, 2024-02-23:
Error scrolling again.  Tried disabling the first tuner in the list this time and sdrtrunk closed.

bradjtrammell, 2024-03-01:
> USB transfer buffer failed

This error usually means that you're out of USB bandwidth, and that your system is halting the software to prevent issues.

- What operating system is this? (Windows, Linux, Mac)
- What are you running SDR Trunk on? (eg Laptop or Desktop)
- How many SDR's are you using?

n5lbr, 2024-03-01:
Hi Brad,  

It's an Intel NUC 13 desktop running Windows 11 pro (64 bit, build 22631.3155).  The CPU is an i7 with 16 cores and 64 GB RAM.  The two Airspy dongles are connected to the USB ports on the motherboard (no USB hub).

OhioRSN, 2024-05-04:
I am having this issue but notice it only when setting preferred tuners or SDR trunk cannot lock onto the control channel. Currently monitoring multiple sites and some VHF and a DMR channel.
14 Nooelec SDR Tuners (3 Powered USB Hubs)
24 Core I9 13900K
64 GB DDR5 RAM (48 Allocated To SDR Trunk)
[image]
[image]
[image]

PR #1950 (2024-08-21): 1860 LibUsb Transfer Buffer Error Handling Enhancements
Closes #1860 
Resolves USB transfer buffer (re)submit error tracking. Synchronizes the submit transfers method so that multi-threaded access doesn't cause accounting errors. Ignores LIBUSB_ERROR_BUSY when (re)submitting due to accounting errors. 

Removes several P25P1 unrecognized messaging loggers.
closed 2024-08-21 by DSheirer