DSheirer/sdrtrunk #1861: Unable to Stream to rdio-scanner via API on v0.6.1-beta-1

sschwetz, 2024-02-22:
v0.6.1-beta-1
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.

When attempting to use the API to upload to rdio-scanner the connection shows as established, but 
rdio-scanner shows the error in it's logs "api: incomplete call data: no talkgroup"

**To Reproduce**
Steps to reproduce the behaviour:
1. Create Streaming connection to use rdio-scanner API
2. Once connected, no uploads are accepted by rdio-scanner and the above message is logged in the log

**Expected behaviour**
Each recording should be uploaded to rdio-scanner via the API

**Application Log**
Example rdioscanner Log Message
[image]

No errors are found in the SDRTrunk log file

**Desktop (optional - complete the following information):**
 - OS: Windows 11
 - CPU Cores: 8
 - RAM:32

bradjtrammell, 2024-03-01:
This isn't a bug in SDR Trunk.

Your copy of Rdio Scanner is misconfigured. When it receives a call, the talkgroup or talkgroup ID SDR Trunk is sending over, isn't being accepted by Rdio Scanner. 

Check your settings in Rdio Scanner.

grahampaull, 2024-10-15:
Agree with the above, you will need to either Add the TG into RDIO-Scanner or turn on Auto Populate.

closed 2026-11-08 by DSheirer