DSheirer/sdrtrunk #1863: Heap leak

amigan, 2024-02-28:
**sdrtrunk Version**
master-branch

**Describe the bug**
I noticed on a couple occasions my two tuner raspberry pi 5 8GB would spin its fan up with occasional truncated calls, and suspected GC hell. I started using a JMX prometheus exporter to take a look. It haven't gotten it reproduce that specific behavior yet, but heap usage is definitely increasing over the long term.
[image]
[image]

The middle of those graphs was a restart. I am fairly certain the blip of concurrent GC was while the system was otherwise loaded. I am continuing to leave this running to collect more data, but it definitely looks like the heap is growing? I'm not a java guy, so forgive me.

I am happy to take any dumps, or enable further profiling to assist in the effort to fix this.

**To Reproduce**
Steps to reproduce the behavior:
1. Monitor the RISCON P25 system (10MHz allocation, 4x simulcast sites) with two RTL-SDRs.
2. Leave it running.

**Expected behavior**
Not fans spinning up with calls truncated.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
- Raspberry Pi 5 8GB running Arch Linux ARM and liberica JDK
- Two RTL-SDRs 2.4MSPS

**Additional context**
Streaming to rdio-scanner running on another machine and an icecast sink. This and icecast-kh are the sole workload of this system. The P25 system is moderately busy at times but not insane. I do think the problem manifests quicker during rare traffic spikes.

amigan, 2024-03-04:
This is definitely happening right now. I have taken a heap dump hprof that I can furnish if necessary.

[image]
[image]

I have attached a heap histogram.
[histogram.txt](https://github.com/DSheirer/sdrtrunk/files/14482202/histogram.txt)

DSheirer, 2024-10-27:
@amigan can you please run your testing against the latest nightly and let me know if you're seeing the same behavior.

The latest nightly is running JDK23 which has a different GC profile.

There will certainly be some level of increase of heap usage over time as the application stores/caches some of the network details but it should be more than a couple tens of megabytes.

If you perform a manual garbage collect, does the heap usage reduce in size? 

Also, are you looking specifically at allocated heap, or used heap?  VisualVM can give you details for those 2 values.

amigan, 2024-10-28:
I've installed current `HEAD` and jdk23 and will let things run a few days to see if anything changes. I have never tried to manually initiate GC, but will do so if this issue recurs. Those graphs do specifically show used heap.

amigan, 2024-10-30:
@DSheirer Things are looking promising. I will continue to monitor.

[image]

amigan, 2024-11-04:
@DSheirer Unfortunately the issue is still occurring. A manual GC does nothing, and left long enough, it indeed begins to truncate calls.

[image]

DSheirer, 2024-11-04:
Can you capture a Heap Dump in Visual VM and send it to me?  I'm interested to see what's being held onto and not garbage collected.

DSheirer, 2024-11-04:
To clarify, can you capture a Heap Dump when it's in that extreme memory usage state where the GC is not bring the consumed heap back down to minimums.

amigan, 2024-11-05:
I built the latest HEAD and am monitoring. I will take a heap dump once it enters the state.

amigan, 2024-11-06:
@DSheirer I have a heap dump. It is 6GB; how should I get it to  you?

DSheirer, 2024-11-08:
Heap dump showed memory leak in the incoming audio segment queue where the processing thread either died (unlikely) or was never started (better possibility).  Redesigned the thread start/stop here #2086 and will keep this issue open until testing shows if the issue is resolved.

amigan, 2024-11-10:
This looks like it is fixed. Thank you!
[image]

DSheirer, 2024-11-10:
Awesome!  Thanks for you help in capturing that heap dump.  I don't think I would have been able to chase that down without it.

closed 2024-11-10 by DSheirer