DSheirer/sdrtrunk #1868: SDRTrunk v0.6.1-beta-1

vboeree, 2024-03-04:
Hi, I'm using SDRTrunk v0.6.1-beta-1 with an SDRPlay RSP1a. The sample rate is 6 Mhz. The desktop computer is as follows...

Processor:	AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx     2.10 GHz
Installed RAM:	12.0 GB (9.92 GB usable)
System type:	64-bit operating system, x64-based processor

Edition:	Windows 10 Home
Version:	22H2
OS build:	19045.4046
Experience:	Windows Feature Experience Pack 1000.19053.1000.0

Along with the control channel, I set up a dummy channel mid range.
After running sdr-trunk.bat things boot up just fine. After some time passes I get the following...

2024-03-04 17:24:18.200 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [21464763365] from preferred tuner [RSP1A SER:200605A397] - searching for another tuner  [183MB/370MB 49%]
2024-03-04 17:36:25.000 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [8471502040] from preferred tuner [RSP1A SER:200605A397] - searching for another tuner  [253MB/370MB 68%]
2024-03-04 17:36:25.001 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [8448682040] from preferred tuner [RSP1A SER:200605A397] - searching for another tuner  [253MB/370MB 68%]

The lowest frequency is 851.0875 Mhz..
The highest frequency is 853 .5250 Mhz..

Any ideas what the cause of this could be? Thanks, Vendel/ K2DSI.

sschwetz, 2024-03-07:
@vboeree, these 3 channels you have provided are outside the frequencies that you are covering e.g. 

Unable to source channel [8471502040] = 847.150204Mhz

vboeree, 2024-03-07:
The thing I don't understand is why are these frequencies popping up. They aren't part of the trunked system, especially 217.64763365 Mhz. I tried different sample rates and am currently using the 10 Mhz. (8.5 Mhz. usable). I stopped using dummy channels and now only have the control channel entered. I think I may go to the latest stable version and give that a try. :)

closed 2024-03-10 by vboeree