DSheirer/sdrtrunk #1871: Suggestion: Set Priority Channel for Duplicate Calls

JamesK455, 2024-03-08:
**Situation:**
I am located within listening range of 4 P25 sites that are all part of one provincial system (not simulcast). As they're part of the provincial trunked system the same talkgroup is often active on more than one of the sites -- but not always. 2/4 of the control traffic channels are at the outer limit of my reception range so the audio isn't always as clear as the site that is a block away. I sometimes have recordings that are patchy/garbled due to weak decodes but when I watch the channel activity the same audio was available from the stronger site.

**Describe the solution you'd like**
It would be great if, in addition to the duplicate call detection that is currently working exactly as advertised, I could select priority channels. 

Basically: if there is a duplicate call, prefer *{channel}* otherwise use whatever is available.

**Describe alternatives you've considered**
I considered separate alias lists per channel even though the talkgroups and radio IDs are shared throughout the system but this approach caused missed calls when the departments are all connected to a different site than the one I have set to record. 

I also tried turning off the duplicate detection but this became difficult to manage due to the volume of calls I am streaming/recording.

**Additional context**
The program is awesome and I appreciate the work put into it! This is by no means a dealbreaker for me but would make monitoring slightly easier during high call volume events and mutual aid situations.

thisisfakediy, 2024-03-14:
I would love this feature as well.

leoninelion, 2024-03-14:
Error counts per channel could be maintained in a table, events like SYNC LOSS, etc would count to a total "channel quality" score for the control channel.  If assumed that every channel spawned by that control channel would have that same quality score, automatic channel quality voting could occur.

ArrEssJay, 2024-04-11:
I would suggest tracking the short term moving average of voice frame bit-error-rates on a per-traffic-channel basis for each site, then feeding the decoder from the least erroneous source as the call progresses, noting that as the call may hop between traffic channels and observed channel error rate at the receiver can change abruptly, there could be a need to switch source mid-call.

For example, I have a receiver located such that co-channel interference is a problem. Some traffic/control channels are used by multiple sites with similar received signal strength. The likelihood of trunking/traffic being received from a given site is predominantly a function of site distance, time of day (system utilisation) and channel allocation/re-use between the sites.

LCEbay, 2024-07-06:
I, too, would like a feature like this.  I can receive 3-5 sites (depending on conditions).  Any given call will probably appear on 2 or 3 of the sites.  One of the sites has traffic that does not appear on the other sites, but of course, is fringe for me.  Any duplicate calls that I receive on that fringe site, can be discarded, as they are surely of lesser quality than the sites that are closer to me.

I think @leoninelion and @ArrEssJay make some valid points, that tracking errors would also be a great approach, and arguably better.  But I'm also not one to let perfect be the enemy of good.