DSheirer/sdrtrunk #1874: Lost P25 P1 control channels in multi-control channel playlist

leoninelion, 2024-03-14:
**sdrtrunk Version**
0.6.1-beta-1

**Configuration Environment**
4 Nooelec RTL-SDR listening to 5 P25 P1 control channels for linked system

**Describe the bug**
After an unknown amount of time (can be hours to weeks) 4 of the 5 defined control channels will revert to IDLE, while one channel remains CONTROL.  IDLE channel Messages indicate SYNC LOSS and the Channel graph is flat, showing no RX signal presence.  Restarting the channels has no effect. The Tuners display shows all tuners Enabled and still locking frequencies (non-zero (LOCKED) displayed), suggesting all SDRs are still functioning.  Reception of voice channels on affected control channels stops system-wide, including on Now Playing and streams.  The remaining control channel still shows CONTROL and Channel graph shows normal signal presence and streaming.  Resolution is restarting the application.  

**Screenshots**

[image]
[image]

**Application Log**
Regular messages of the type "Error deleting temporary internet recording file" occur regularity, but are not coincident with this problem.  No other errors exist at the time of these problems.

**Desktop (optional - complete the following information):**
 - OS: Ubuntu 22 x64
 - CPU Cores: 4
 - RAM: 8GB

**Additional context**
This problem has been occurring in all versions 0.6.0, including pre-release.  0.5x was never used.

DSheirer, 2024-10-27:
@leoninelion if you have a chance, can you please test against the latest 0.6.1 Beta 2 release, or against the nightly release and let me know if you still see this issue?

leoninelion, 2024-11-04:
Ok, I went to B3 and will be monitoring, I'll report back if it acts up.  Thank you very much.

leoninelion, 2024-11-04:
I can already see it's stopped giving me "Error deleting temporary internet recording file:" errors now. 👍

DSheirer, 2024-11-04:
Ok.  Please let me know if it's good and we can close this ticket.  Thanks!

leoninelion, 2024-12-19:
Ok, I've not seen this happen since the new code, this is ok to close.  Thank you very much!

closed 2024-12-19 by DSheirer