DSheirer/sdrtrunk #1875: 0.6.1 b1 - Threading Deadlock

kazzaw, 2024-03-18:
**sdrtrunk Version**
Utilizing 0.6.1 beta 1 

Log attached
[20240318_112438_sdrtrunk_processing_diagnostic_report.log](https://github.com/DSheirer/sdrtrunk/files/14630711/20240318_112438_sdrtrunk_processing_diagnostic_report.log)

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
Virtual Machine - Server 2019
10GB RAM, 8600t Processor

DSheirer, 2024-03-18:
@kazzaw this issue is likely fixed in the nightly build and will be included in the next release.

If you could test using the nightly release to see if you're still having this threading deadlock issue and let me know if the issue still persists?

DSheirer, 2024-03-18:
Note: diagnostic report log indicates build timestamp is 10Feb2024 and DMR threading deadlock issue was resolved in commit f5938fa8e2aa4f389de60b962104f02a04918ca3 on 13Feb2024

kazzaw, 2024-03-18:
Thanks Denny, I’ll deploy the nightly build tonight and monitor. It usually takes a couple days to rear its head, but if you don’t hear further from me within the week you can consider this resolved

kazzaw, 2024-03-28:
As described above, running the nightly has resolved. Look forward to seeing in b2

closed 2024-03-28 by kazzaw