DSheirer/sdrtrunk #1880: 2024-03-26 11:19:21.979 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [10297945700] from preferred tuner...

vboeree, 2024-03-26:
Hi I'm running sdrtrunk nightly - 2025-02-13t09:27:56.887+0000 with my RSP1A and the latest API, V3.14. I have the sample rate set wide enough for the P25-1 system I'm monitoring. After listening for a while I get the following message,

2024-03-26 11:19:21.979 INFO  i.g.d.s.t.m.TunerManager - Unable to source channel [10297945700] from preferred tuner [RSP1A SER:200605A397] - searching for another tuner  [548MB/1GB 52%]

Any ideas? I didn't have this problem with the latest stable version and the prior API?

DSheirer, 2024-03-27:
The frequency value (10297945700 = 10,297.945700 MHz) is invalid ... there are no trunked systems operating in the 10 GHz frequency range.

One of the decoded messages, specifically a channel grant message, would have produced this value.  In order to diagnose, I'll need to know which message produced it.

I'll need a bitstream recording of the control channel from the same time that this message is produced and then I can review the bitstream to find out which message is producing this bad frequency value.

vboeree, 2024-03-27:
Hi, thanks for getting back to me so quickly and also thanks for the time and hard work you put into this software. I'll do my best to get that Info. to you once it occurs again, Vendel.

vboeree, 2024-03-28:
How do I get a bitstream of the control channel? Do I use one of the recording options? Sorry about all the questions, Vendel.

DSheirer, 2024-03-28:
yes, in the channel configuration editor, in the Recording tab at the bottom, turn on the control channel bistream record option and then start the channel.

It will record up to 512kb per recording and automaticaly create a new one once it gets to that size, and so you only need to send the recording that covers the time period for when you see this error message happen.

vboeree, 2024-03-28:
I hope this is correct. :D

Oh no. it won't let me attach a wav file. Hmmm...

vboeree, 2024-03-28:
GitHub wouldn't let me send these files. Hopefully I can send it this way, Vendel.

vboeree, 2024-04-04:
I replaced the "SDRTrunk - Playlist Editor Channels" entry and included the alternate control channels. The issue hasn't happened since then. We'll keep an eye on it...

closed 2024-10-27 by DSheirer