DSheirer/sdrtrunk #1881: Api error -no frequency table

scottyb908, 2024-03-27:
**sdrtrunk Version**

**Describe the bug**

I am trying to get the SDR Trunk set up to feed to broadcatify calls from a conventional system. Whenever there's activity I am seeing the stream being uploaded to the standard streams but calls doesn't report any activity. in the logs I am seeing an error [Broadcastify Calls API upload URL Request failed [1 no-freq-table-entry-for-upload]. I don't see any place to add freq information specific to calls.

**To Reproduce**
Steps to reproduce the behavior:
Create a calls broadcastify system, 
configure tuner and create “talk group”/alias
Try to upload transmission to broadcastify. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: [e.g. iOS]
 - CPU Cores: [e.g. 4 cores]
 - RAM:[e.g. 16GB]

**Additional context**
Add any other context about the problem here.

DSheirer, 2024-03-27:
I don't know what this error means.  Have you reported this to the Broadcastify support forum?

scottyb908, 2024-03-27:
I have posted there as well. I am not sure which side is causing the error, bad api response vs bad call. Sent from my iPhone

closed 2024-03-27 by DSheirer