DSheirer/sdrtrunk #1884: Radio ID alias

minecrafter12345, 2024-03-29:
A good idea would be to make the Radio Alias display instead of the Radio ID, when for instance it changes channel, or affiliates to whatever channel its on.

I would like for the Radio Alias, and the channel Alias to pop up when a radio changes channel or affiliates with a channel instead of the channel TG Number or Radio ID number.
This us under the Events Tab.
Thanks! hopefully this made sense lol

ikm19, 2024-09-02:
you already can do that. 

open your playlist editor

go to tab "Aliases" then to secondary tab "Alias"

on the right click "new"

this will create new entry in the alias list. then bellow the list add your name, group and identifiers as on the screenshot bellow. hope this help my friend. happy scanning.
[image]

minecrafter12345, 2024-09-02:
Yes, I have those setup, I am refering in the events tab, where you can watch and see what channel a radio swaps to, on the side instead of the channel name it’s uses the number instead of the alias for that channel

DSheirer, 2024-10-27:
This sounds very comlicated.  Closing.

closed 2024-10-27 by DSheirer