DSheirer/sdrtrunk #1886: Crashing without reason

tweiss3, 2024-04-05:
v0.6.1 Beta 1

I updated on 2/10/24 to this version. Since then, the program is crashing without notice every 7-10 days. I have since started a script that will restart the program if it notices the crash (restarted twice in 2 weeks), so the crashes are still happening.

Today I got a notice the node was offline. It dropped the SDR (Airspy R820T), and wouldn't recognize it until I restarted the program.
 
Previous v0.5.0 Alpha 6 usually went 6-8 weeks before an issue.

OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Manufacturer	Dell Inc.
System Model	Precision T1700
System Type	x64-based PC
System SKU	05A6
Processor	Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz, 3601 Mhz, 4 Core(s), 8 Logical Processor(s)
Installed Physical Memory (RAM)	16.0 GB
Total Physical Memory	15.9 GB
Available Physical Memory	5.09 GB
Total Virtual Memory	19.8 GB
Available Virtual Memory	2.64 GB
Page File Space	3.94 GB

Current system utilization:
[image]

System is currently only running 1 SDRTrunk and 2 ProScan (zero issues with proscan running nostop).

DSheirer, 2024-05-04:
@tweiss3 I need more details to understand the issue.  What do you mean by 'crash'?  Does the application lock up and become unresponsive, does it run out of memory?   There may be errors in the application log ... can you post the application log after it crashes?

tweiss3, 2024-05-13:
@DSheirer 

I guess it is two seperate issues:
1) I would get a notification that the node was offline, and when logging on this computer (this is a dedicated computer to streaming radio), the application had crashed completely and closed out. No instances running, nothing, just the desktop. 

After noticing the complete crashes a few times, I found a program called Restart On Crash, that would restart SDRtrunk after the crash, and that worked for a few weeks, until I started getting errors for the Airspy SDR.

2) The Airspy SDR errors when looking at the tuner was "bus transfer buffers exhausted". I could restart it, but anywhere between 5 minutes and 90 minutes the error would return. This wasn't a "crash" so it wouldn't restart automatically.

I looked back, all the logs have since been overwritten (thought it kept logs longer than a few days, but I was mistaken).

Since I listen to these streams every day, and couldn't find a work around the SDR errors, I have since rolled back to v0.5.0 alpha6, and it has been running without error since 4/6 (1.46 millions calls uploaded).

DSheirer, 2024-10-27:
@tweiss3 have you tested with the latest nightly release or the 0.6.1 beta 2 release to see if this issue is still present?

closed 2026-11-08 by DSheirer