DSheirer/sdrtrunk #1889: SDRPlay RSP1B tuner settings

dleiken, 2024-04-17:
The tuner settings for the SDRPlay RSP1B are very limited compared to a tuner like the Airspy. Will more options be available in the future?

DSheirer, 2024-04-17:
I can only give you control over the settings that the tuner vendor provides.  Is there a specific setting that's available via the SDRPlay API for the RSP1B that you know is missing?

dleiken, 2024-04-17:
As you can see from the attached photo I need to make some adjustments for proper decoding. When I turn down the gain it loses locking on my CC and when gain is up it does not decode 100%. Should I be making settings change with the SDRPlay console, and will they reflect to SDRT? I don’t see any settings within the API. Thanks for your help. Sent from my iPhone

DSheirer, 2024-05-04:
@dleiken there's no photo attached here.  You'll need to use the github interface to attach photos.

You should only make changes in the sdrtrunk application ... I don't know/think that making changes to an SDRPlay tuner that is in-use by sdrtrunk will have any effect.

The user manual for setting gain on sdrplay tuners is here: https://github.com/DSheirer/sdrtrunk/wiki/Tuners#setting-gain-on-rsp-tuners

culleydog2005, 2024-08-01:
Polk County Florida scanner for all sites down

closed 2024-10-27 by DSheirer