DSheirer/sdrtrunk #1893: Can't login to RadioReference - userInfo is Null

haruny, 2024-04-21:
**sdrtrunk Version**
0.6.1-beta.1

**Describe the bug**
Radio Reference doesn't work for a premium user. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Playlist Editor
2. Click on Radio Reference
3. Click Login
4. Enter user name password
5. click test connection
6. See error in console

**Expected behavior**
Login is expected.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**

Exception in thread "JavaFX Application Thread" java.lang.NullPointerException: Cannot invoke "io.github.dsheirer.rrapi.type.UserInfo.getExpirationDate()" because "ui" is null
        at io.github.dsheirer.service.radioreference.RadioReference.testConnectionWithExp(RadioReference.java:191)
        at io.github.dsheirer.gui.playlist.radioreference.LoginDialog$2$1.run(LoginDialog.java:323)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$11(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication._enterNestedEventLoopImpl(Native Method)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication._enterNestedEventLoop(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.Application.enterNestedEventLoop(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.EventLoop.enter(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.enterNestedEventLoop(Unknown Source)
        at javafx.graphics/javafx.stage.Stage.showAndWait(Unknown Source)
        at javafx.controls/javafx.scene.control.HeavyweightDialog.showAndWait(Unknown Source)
        at javafx.controls/javafx.scene.control.Dialog.showAndWait(Unknown Source)
        at io.github.dsheirer.gui.playlist.radioreference.RadioReferenceEditor.lambda$getLoginButton$10(RadioReferenceEditor.java:532)
        at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
        at javafx.base/javafx.event.Event.fireEvent(Unknown Source)
        at javafx.graphics/javafx.scene.Node.fireEvent(Unknown Source)
        at javafx.controls/javafx.scene.control.Button.fire(Unknown Source)
        at javafx.controls/com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(Unknown Source)
        at javafx.controls/com.sun.javafx.scene.control.inputmap.InputMap.handle(Unknown Source)
        at javafx.base/com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(Unknown Source)
        at javafx.base/com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(Unknown Source)
[... 57 lines cut ...]
        at javafx.base/com.sun.javafx.event.EventUtil.fireEventImpl(Unknown Source)
        at javafx.base/com.sun.javafx.event.EventUtil.fireEvent(Unknown Source)
        at javafx.base/javafx.event.Event.fireEvent(Unknown Source)
        at javafx.graphics/javafx.scene.Scene$MouseHandler.process(Unknown Source)
        at javafx.graphics/javafx.scene.Scene.processMouseEvent(Unknown Source)
        at javafx.graphics/javafx.scene.Scene$ScenePeerListener.mouseEvent(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(Unknown Source)
        at javafx.graphics/com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.View.handleMouseEvent(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.View.notifyMouse(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)

**Desktop (optional - complete the following information):**
 - OS: Windows 11
 - CPU Cores: 24
 - RAM: 32GB

**Additional context**
Add any other context about the problem here.

DSheirer, 2024-04-21:
This seems like an issue with the radio reference API.  I'll leave open until this is resolved.

DSheirer, 2024-05-04:
This should be resolved now.  Closing.

closed 2024-05-04 by DSheirer