DSheirer/sdrtrunk #1895: Transition Radio Reference API To Use RPC/Encoded

DSheirer, 2024-04-22:
The RR client code currently uses doc/lit style of API method calls and needs to migrate to rpc/encoded format.  RR is moving to no-longer support the doc/lit format as of April 2024.

bgAD666, 2024-04-23:
Hello
SDR Trunk
Is it possible to create a multilingual font, or follow the fonts used by the computer?
Since English is not my mother tongue
Thanks

DSheirer, 2024-04-24:
> Hello SDR Trunk Is it possible to create a multilingual font, or follow the fonts used by the computer? Since English is not my mother tongue Thanks

Which language do you need?

bgAD666, 2024-04-24:
I need Chinese language
Or is it possible to create a font file
And then I do the translation myself?
Thanks