DSheirer/sdrtrunk #1902: Unrecognized LCW Opcode

rxes, 2024-05-03:
**sdrtrunk Version**
Nightly from April 27

**Describe the bug**
Keep receiving Unrecognized LCW Opcode

**To Reproduce**
Start the application

**Expected behavior**
The previous nightly version didn't exhibit this problem

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
20240503 080658.275 [sdrtrunk cached thread 1] INFO  i.g.d.g.p.d.JmbeLibraryPreferenceEditor - Current: 1.0.9  [69MB/258MB 26%]
20240503 080658.276 [sdrtrunk cached thread 1] INFO  i.g.d.g.p.d.JmbeLibraryPreferenceEditor - Available: 1.0.9  [70MB/258MB 27%]
20240503 080658.277 [sdrtrunk cached thread 1] INFO  i.g.d.g.p.d.JmbeLibraryPreferenceEditor - No JMBE library update is available at this time  [71MB/258MB 27%]
20240503 080743.718 [main] INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : nightly  [9MB/256MB 3%]
20240503 080743.719 [main] INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.2  [9MB/256MB 3%]
20240503 080743.719 [main] INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2024-04-27T14:37:05.354+0000  [9MB/256MB 3%]
20240503 080743.720 [main] INFO  i.g.d.log.ApplicationLog - Build-JDK         : 20.0.2 (BellSoft 20.0.2+10  [9MB/256MB 3%]
20240503 080743.720 [main] INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.5.0-1018-azure  [9MB/256MB 3%]
20240503 080743.720 [main] INFO  i.g.d.log.ApplicationLog -   [9MB/256MB 3%]
20240503 080743.721 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/256MB 3%]
20240503 080743.721 [main] INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [9MB/256MB 3%]
20240503 080743.721 [main] INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [9MB/256MB 3%]
20240503 080743.721 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [9MB/256MB 3%]
20240503 080743.722 [main] INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [9MB/256MB 3%]
20240503 080743.722 [main] INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 10  [9MB/256MB 3%]
20240503 080743.722 [main] INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [9MB/256MB 3%]
20240503 080743.723 [main] INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [9MB/256MB 3%]
20240503 080743.723 [main] INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        8  [9MB/256MB 3%]
20240503 080743.723 [main] INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  3 GB  [9MB/256MB 3%]
20240503 080743.724 [main] INFO  i.g.d.log.ApplicationLog - Storage Directories:  [9MB/256MB 3%]
20240503 080743.724 [main] INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\rxone\SDRTrunk  [9MB/256MB 3%]
20240503 080743.725 [main] INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\rxone\SDRTrunk\logs  [9MB/256MB 3%]
20240503 080743.726 [main] INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\rxone\SDRTrunk\event_logs  [9MB/256MB 3%]
20240503 080743.726 [main] INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\rxone\SDRTrunk\playlist  [9MB/256MB 3%]
20240503 080743.727 [main] INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\rxone\SDRTrunk\recordings  [9MB/256MB 3%]
20240503 080743.744 [main] INFO  i.g.dsheirer.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [9MB/256MB 3%]
20240503 080743.753 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\rxone\SDRTrunk\SDRTrunk.properties]  [9MB/256MB 3%]
20240503 080744.074 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [17MB/34MB 51%]
20240503 080744.250 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.262  [18MB/34MB 54%]
20240503 080744.252 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.22.11312  [18MB/34MB 54%]
20240503 080744.475 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [16] potential usb devices  [18MB/34MB 54%]
20240503 080744.483 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [2] Port [1.2] Tuner Class [Airspy HF+]  [19MB/34MB 55%]
20240503 080744.487 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - Airspy HF+ USB Bus:2 Port:1.2 - Added / Starting ...  [19MB/34MB 55%]
20240503 080744.668 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [912000.0] providing [36] channels at [25333.3] Hz each  [23MB/34MB 67%]
20240503 080744.672 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [23MB/34MB 67%]
20240503 080744.683 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [23MB/34MB 69%]
20240503 080744.756 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [8MB/34MB 26%]
20240503 080744.764 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library v3.14 - loaded  [9MB/34MB 27%]
20240503 080744.802 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered [2] RSP devices from SDRplay API  [10MB/34MB 29%]
20240503 080744.804 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: RSPduo Tuner 1 SER#2305003934 - Added / Starting ...  [10MB/34MB 29%]
20240503 080744.886 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [1500000.0] providing [60] channels at [25000.0] Hz each  [12MB/34MB 36%]
20240503 080744.892 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: RSP1A SER#2303006398 - Added / Starting ...  [12MB/34MB 37%]
20240503 080744.913 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [1500000.0] providing [60] channels at [25000.0] Hz each  [12MB/34MB 37%]
20240503 080744.922 [main] INFO  i.g.d.settings.SettingsManager - SettingsManager - loading settings file [C:\Users\rxone\SDRTrunk\settings\settings.xml]  [12MB/34MB 38%]
20240503 080745.039 [main] INFO  i.g.d.m.DiagnosticMonitor - Diagnostic monitoring enabled running every 30 seconds  [18MB/34MB 53%]
20240503 080745.349 [main] WARN  i.g.d.vector.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 4, S_128_BIT]  [12MB/34MB 37%]
20240503 080746.100 [main] INFO  i.g.d.playlist.PlaylistManager - Loading playlist [C:\Users\rxone\SDRTrunk\playlist\default.xml]  [26MB/34MB 76%]
20240503 080747.106 [main] INFO  i.github.dsheirer.gui.SDRTrunk - starting main application gui  [28MB/148MB 19%]
20240503 080748.622 [sdrtrunk scheduled thread 2] INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] now - timer invoked  [52MB/148MB 35%]
20240503 080748.700 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.JmbeAudioModule - Loading JMBE library from [C:\Users\rxone\SDRTrunk\jmbe\jmbe-1.0.9.jar]  [56MB/148MB 38%]
20240503 080748.710 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [57MB/148MB 38%]
20240503 080748.710 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [57MB/148MB 38%]
20240503 080748.734 [sdrtrunk scheduled thread 2] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [1500000.0] providing [60] channels at [25000.0] Hz each  [58MB/148MB 39%]
20240503 080752.021 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:1 MSG:81A4110DFFFFFFFFFFFFF79D [Log Suppress 1/1]  [51MB/148MB 35%]
20240503 080752.027 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:15 MSG:8FA40A80000A00000000B5F0 [Log Suppress 1/1]  [54MB/148MB 36%]
20240503 080822.932 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [912000.0] providing [36] channels at [25333.3] Hz each  [102MB/214MB 48%]
20240503 080823.820 [sdrtrunk channel [2/P25-1] 155242500 thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized LCW Opcode: L3HARRIS_UNKNOWN VENDOR:HARRIS OPCODE:50 MSG:32A444414453503036 CHAN:null FREQ:155242500 [Log Suppress 1/1]  [65MB/214MB 30%]
[... cut ...]
20240503 080847.067 [sdrtrunk channel [4/P25-1] 154972500 thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized LCW Opcode: L3HARRIS_UNKNOWN VENDOR:HARRIS OPCODE:53 MSG:35A44F484E534F4E00 CHAN:null FREQ:154972500 [Log Suppress 1/1]  [123MB/214MB 57%]
20240503 080853.670 [sdrtrunk channel [4/P25-1] 154972500 thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized LCW Opcode: L3HARRIS_UNKNOWN VENDOR:HARRIS OPCODE:51 MSG:33A463310000000000 CHAN:null FREQ:154972500 [Log Suppress 1/1]  [37MB/214MB 17%]
**Additional context**
Add any other context about the problem here.

rxes, 2024-05-30:
I just download the new nightly and there is still decoding issues.

20240530 071458.288 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [18MB/4GB 0%]
20240530 071458.492 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.262  [20MB/4GB 0%]
20240530 071458.493 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.22.11312  [20MB/4GB 0%]
20240530 071458.714 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [16] potential usb devices  [21MB/4GB 0%]
20240530 071458.721 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [2] Port [1.2] Tuner Class [Airspy HF+]  [21MB/4GB 0%]
20240530 071458.725 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - Airspy HF+ USB Bus:2 Port:1.2 - Added / Disabled  [21MB/4GB 0%]
20240530 071458.728 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [21MB/4GB 0%]
20240530 071458.739 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [21MB/4GB 0%]
20240530 071458.843 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [24MB/4GB 0%]
20240530 071458.852 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library v3.14 - loaded  [24MB/4GB 0%]
20240530 071458.893 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered [2] RSP devices from SDRplay API  [25MB/4GB 0%]
20240530 071458.914 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: RSPduo Tuner 1 SER#2305003934 - Added / Starting ...  [27MB/4GB 0%]
20240530 071459.021 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [3000000.0] providing [120] channels at [25000.0] Hz each  [31MB/4GB 0%]
20240530 071459.027 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: RSP1A SER#2303006398 - Added / Starting ...  [31MB/4GB 0%]
20240530 071459.050 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [3000000.0] providing [120] channels at [25000.0] Hz each  [31MB/4GB 0%]
20240530 071459.058 [main] INFO  i.g.d.settings.SettingsManager - SettingsManager - loading settings file [C:\Users\rxone\SDRTrunk\settings\settings.xml]  [32MB/4GB 0%]
20240530 071459.192 [main] INFO  i.g.d.m.DiagnosticMonitor - Diagnostic monitoring enabled running every 30 seconds  [37MB/4GB 0%]
20240530 071459.820 [main] WARN  i.g.d.vector.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 4, S_128_BIT]  [49MB/4GB 1%]
20240530 071500.690 [main] INFO  i.g.d.playlist.PlaylistManager - Loading playlist [C:\Users\rxone\SDRTrunk\playlist\default.xml]  [33MB/4GB 0%]
20240530 071501.757 [main] INFO  i.github.dsheirer.gui.SDRTrunk - starting main application gui  [77MB/4GB 1%]
20240530 071503.306 [sdrtrunk scheduled thread 2] INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] now - timer invoked  [103MB/4GB 2%]
20240530 071503.388 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.JmbeAudioModule - Loading JMBE library from [C:\Users\rxone\SDRTrunk\jmbe\jmbe-1.0.9.jar]  [106MB/4GB 2%]
20240530 071503.414 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [106MB/4GB 2%]
20240530 071503.414 [sdrtrunk scheduled thread 2] INFO  i.g.d.a.c.mbe.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [106MB/4GB 2%]
20240530 071503.434 [sdrtrunk scheduled thread 2] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [3000000.0] providing [120] channels at [25000.0] Hz each  [107MB/4GB 2%]
20240530 071506.869 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:1 MSG:81A4110DFFFFFFFFFFFFF79D [Log Suppress 1/1]  [976MB/4GB 23%]
20240530 071506.961 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:15 MSG:8FA40A80000A00000000B5F0 [Log Suppress 1/1]  [1GB/4GB 25%]
20240530 071602.969 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [3000000.0] providing [120] channels at [25000.0] Hz each  [1GB/4GB 46%]
20240530 071636.911 [sdrtrunk channel [0/P25-1] MULTI FREQ thread 1] ERROR i.g.d.sample.Broadcaster - Error while broadcasting [class io.github.dsheirer.module.decode.p25.phase1.message.pdu.ambtc.osp.AMBTCStatusQuery] to listeners [Log Suppress 1/5]  [2GB/4GB 52%]
java.lang.NullPointerException: Cannot invoke "io.github.dsheirer.module.decode.p25.phase1.message.pdu.block.UnconfirmedDataBlock.getMessage()" because the return value of "io.github.dsheirer.module.decode.p25.phase1.message.pdu.ambtc.osp.AMBTCStatusQuery.getDataBlock(int)" is null
	at io.github.dsheirer.module.decode.p25.phase1.message.pdu.ambtc.osp.AMBTCStatusQuery.getSourceAddress(AMBTCStatusQuery.java:90)
	at io.github.dsheirer.module.decode.p25.phase1.message.pdu.ambtc.osp.AMBTCStatusQuery.getIdentifiers(AMBTCStatusQuery.java:108)
	at io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:58)
	at io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:43)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
[... cut ...]
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)

The nightly from 2-13-24 is working properly.

ms4sman, 2024-05-30:
I am also seeing these errors in my log now, after updating from 0.6.0 to the latest nightly from a few hours ago. I am not sure what they mean though.

bhall7, 2024-06-21:
I'm running the SDRTrunk nightly build from June 2 and monitoring a new statewide P25 Phase 2 system in SDRTrunk that is coming online here in Utah (https://www.radioreference.com/db/sid/12042), and I'm seeing some similar "Unrecognized TSBK Opcode" errors in the log:

```
0240611 162111.344 [sdrtrunk channel [1/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:1 MSG:81A4110DFFFFFFFFFFFFF79D [Log Suppress 1/1]  [108MB/366MB 29%]
20240611 162111.389 [sdrtrunk channel [1/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: HARRIS_OSP_UNKNOWN VENDOR:HARRIS OPCODE:15 MSG:8FA40A82000A00000000D513 [Log Suppress 1/1]  [133MB/366MB 36%]
2024
0611 163052.140 [sdrtrunk channel [1/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: UNKNOWN_VENDOR_OSP VENDOR:V166 OPCODE:36 MSG:A4A6B80BB840AAE3EF1EB32B [Log Suppress 1/1]  [281MB/366MB 76%]
20240611 163122.840 [sdrtrunk channel [1/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: UNKNOWN_VENDOR_OSP VENDOR:V173 OPCODE:52 MSG:B4ADB5F141008082373124C0 [Log Suppress 1/1]  [243MB/366MB 66%]
20240611 163613.689 [sdrtrunk channel [1/P25-1] MULTI FREQ thread 1] INFO  i.g.d.m.d.p.p.P25P1DecoderState - Unrecognized TSBK Opcode: UNKNOWN_VENDOR_OSP VENDOR:SEA OPCODE:18 MSG:92C03565654D6091701AB493 [Log Suppress 1/1]  [139MB/366MB 38%]
2024
```

I played with the gain settings, and I'm getting a good signal and traffic on both an RTL-SDR v3 and an Airspy R2. But, I continue to get errors like these continuously. I'm not sure if it's just because it's a new system that is still in the deployment phase, or something else. I'm attaching the full log to this message in the hopes that it is helpful.

[20240611_sdrtrunk_app.log](https://github.com/user-attachments/files/15931099/20240611_sdrtrunk_app.log)

closed 2024-10-27 by DSheirer