DSheirer/sdrtrunk #1910: Call Audio Aliased As Do Not Monitor Plays Briefly

DSheirer, 2024-05-06:
**Describe the bug**

(as reported on discord)

The latest nightly has an issue where the alias list "listen" setting is not completely honored. That is, aliases without listen checked will still play for roughly half a second before being shut off. Aliases with listen checked work as expected. This is easy to reproduce, especially on busier systems. 

OS MacOS 14.4.1, m1 processor. Have not spun up linux or windows to try to reproduce it there yet. Airspy. Happens on multiple systems, but again easier to notice on busier ones, or it may be dependent on P25 system load. Note, computer is sitting on perhaps 25% cpu.

Does not apply to encrypted calls/TGs - not hearing bursts of encrypted traffic. 

What happens is the audio out is full of less than one second initial parts of muted TG calls, then the audio is suppressed. If unmuted TG traffic occurs, it is handled correctly and the entire call is heard. 

Issue does not occur on earlier released versions, ex 0.5.0 beta 2.

DSheirer, 2024-08-24:
I'm not able to reproduce this issue in local testing on my Linux machine.

closed 2024-08-24 by DSheirer

tadscottsmith, 2024-08-26:
It's not a big deal to me, since I don't listen live, but I can confirm that this occurs for me on MacOS M1 and Windows 10. I can cause it to occur using the same baseband recording provided in #1912.