DSheirer/sdrtrunk #1911: P25 Choppy parts of calls play to speakers while all talkgroups are setup not to play audio

Nokoa, 2024-05-06:
**sdrtrunk Version**
Nightly 5/5/2024

**Describe the bug**
Monitoring multiple P25 Phase 1 control channels with phase 2 voice with all setup with an alias to not play any audio to the speakers.
Small parts of high error rate calls still make it to the speakers. The now playing bar does not show anything when this occurs.

**To Reproduce**
Steps to reproduce the behavior:
Monitor a P25 system with poor reception with an alias to not play any audio for talkgroups with range 1 to 65535. 

**Expected behavior**
All calls should have a talkgroup ID mapped, and if not yet perhaps do not dump audio to the speakers?

**Desktop (optional - complete the following information):**
 - OS: Windows 10
 - CPU Cores:  4 cores
 - RAM: 16GB

Nokoa, 2024-05-06:
Beat me to it #1910

DSheirer, 2024-08-24:
Closing as duplicate to #1910

closed 2024-08-24 by DSheirer