DSheirer/sdrtrunk #1915: No audio playback for P25 with 5/4/24 nightly build

n2ygk, 2024-05-09:
**sdrtrunk Version**
master-branch

**Describe the bug**
Audio for p25 channels does not play for most if not all channels in master nightly build.

**To Reproduce**
Steps to reproduce the behavior:
1. Run sdr-trunk-osx-x86_64-vnightly/bin/sdrtrunk
2. See calls happening in p25 control channel events log
3. No audio heard.
4. Run sdr-trunk-osx-x86_64-v0.6.0/bin/sdrtrunk
5. See calls happening as above
6. audio heard consistently.

**Expected behavior**
Expected to hear audio.

**Screenshots**
If applicable, add screenshots to help explain your problem.
Audion working under v0.6.0:
[image]

Audio not working under vnightly:
[image]

**Application Log**
See vnightly [sdrtrunk_app.log](https://github.com/DSheirer/sdrtrunk/files/15266807/sdrtrunk_app.log)

**Desktop (optional - complete the following information):**
 - OS: macOS Monterey 12.6.4
 - model: MacBook Pro (Retina, 15-inch, Mid 2015)
 - CPU Cores: 2.8 GHz Quad-Core Intel Core i7
 - RAM: 16 GB
 - dongle: RTL-SDR blog v4 RYL2832U/R828D

**Additional context**
Rarely I would hear audio from one of the P25 aliases in the nightly build but noticed most aliases were in the event log but were not playing audio.

I can share the config if needed.

n2ygk, 2024-05-13:
I've also tested with v0.6.1-beta1 and that works OK. I'll try and bisect to find which commit in nightly is the cause.

rxes, 2024-08-21:
Same (for the newest nightly)

DSheirer, 2024-08-24:
> I just installed and have the same issue. User Preferences / Output tones and Test, and I have test sound. Other than that, I can not get any sound from this app. I tried all different frequencies including keying up my radio next to it on same frequency. It shows in waterfall, yet NO sound. Im on windows 11.

@aka1ceman and @rxes have you read the user manual and created a channel configuration?

DSheirer, 2024-08-24:
@n2ygk I'm not sure what would be causing the issue you're seeing.  Have you made any progress in troubleshooting?

n2ygk, 2024-08-24:
@DSheirer sorry no. I got stalled trying to build the components locally. I'm not a Java maven.

rxes, 2024-09-16:
The newest nightly is also not working. I'm attaching .bits  .wav 
[recordings.zip](https://github.com/user-attachments/files/17008307/recordings.zip)

rxes, 2024-09-16:
Still no audio on the latest nightly. Attaching .bits and .wav , hopefully it helps
[image]

DSheirer, 2024-09-19:
I was able to recreate what I think is the same issue.  Audio is playing fine and then all of the sudden it stops playing.  The playback panel still shows the call identifiers and shows that the call is in progress, but no audio plays.  Once it gets into this state it never recovers.  If using stereo playback, both channels will eventually get to this state.

I isolated the issue to the source data line.  When one call ends, the line is drained and closed.  When the next call comes in and we attempt to pre-load audio to fill the line buffer, it fails to write any audio to the line.  I've added a patch to trap this condition and stop/restart the audio line and this seems to be a good recovery.

closed 2024-09-19 by DSheirer

rxes, 2024-09-20:
@DSheirer Same issue with the newest nightly. Not sure what the problem is? It doesn't decode most of the calls (and no sound) unlike the nightly from 2-23, which works great. Adding another recordings zip, hopefully helps 
[recordings.zip](https://github.com/user-attachments/files/17068891/recordings.zip)

n2ygk, 2024-09-20:
@DSheirer just loaded the 9/20/24 nightly and it threw an NPE:
```
java.lang.NullPointerException: Cannot invoke "io.github.dsheirer.module.decode.p25.phase1.P25P1DataUnitID.getMessageLength()" because "this.mDataUnitID" is null
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:350)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:203)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:69)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:87)
	at io.github.dsheirer.dsp.psk.DQPSKGardnerDemodulator.calculateSymbol(DQPSKGardnerDemodulator.java:88)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:146)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:121)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:130)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:39)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:145)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:40)
	at io.github.dsheirer.dsp.filter.channelizer.output.OneChannelOutputProcessor.process(OneChannelOutputProcessor.java:113)
	at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.lambda$new$0(ChannelOutputProcessor.java:56)
	at io.github.dsheirer.util.Dispatcher.process(Dispatcher.java:184)
	at io.github.dsheirer.util.Dispatcher$ProcessorWithHeartbeat.run(Dispatcher.java:226)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
```
Not sure if this is a new thing or just I never noticed.

Waiting for some radio traffic to show up to see if the audio issue is resolved. It's a slow morning on the P25.

n2ygk, 2024-09-20:
@DSheirer I've been running it for about 6 hours now and the audio has continued to work, so I think you've fixed it! Thanks.

DSheirer, 2024-09-21:
> @DSheirer Same issue with the newest nightly. Not sure what the problem is? It doesn't decode most of the calls (and no sound) unlike the nightly from 2-23, which works great. Adding another [...]

@rxes from the 2x recording collections that you posted, it doesn't appear that you are having the same situation/condition as was originally reported by @n2ygk in this issue.  I've opened up a new issue for you here #1989

DSheirer, 2024-09-22:
@rxes I made some changes to the P25 Phase 1 decoder and created test releases and posted them to my google drive.  Could you please download the one for your OS and test to see if this resolves the issue you're seeing and let me know?

Thanks,
Denny

https://drive.google.com/drive/folders/130eCl0Bv-G7RlsBMtIfiP9ZhdCmg_FpL?usp=sharing

rxes, 2024-09-22:
Thank you for getting back to me. I tried it, but it is still the same. 
[recordings 9-22-24.zip](https://github.com/user-attachments/files/17090088/recordings.9-22-24.zip)

anasbousselham, 2025-07-25:
I have the same issue (no audio) with the newest nightly.

PR #1986 (2024-09-19): #1915 (P25) Audio Stops Playing While Calls Continue To Process
Closes #1915 

Audio playback can randomly stop playing audio despite everything else appears to be normal.