DSheirer/sdrtrunk #1919: Issue with the forced 10 MHz spectrum with Airspy R2. Should be manually adjustabled

ItsTechlogix, 2024-05-19:
**sdrtrunk Version**
v0.6.1 beta 1

**Describe the bug**
The software seems to auto select a 10 MHz spectrum to monitor with my Airspy R2. This auto selected spectrum is causing a few of the voice channels to be right at the far edge of the 10 MHz and fade on decode. The tower I am monitoring has a 7.1625 MHz spread between all channels. When I manually change the middle point, it's able to have all the voice channels and the control channel far enough from the edges to produce perfect decodes, but then it auto reverts back to problem spectrum causing the voice channels to be at the edge again. 

Is there anyway to lock the spectrum or add a feature to be able to specify the middle frequency and stay at that without automatically reverting back to a different one?

**To Reproduce**
Steps to reproduce the behavior:
1. Open SDR Trunk with Airspy R2 attached.
2. Set the frequency on the Tuners tab to the middle frequency between the 6 channels on the site being monitored.
3. Start the channel to monitor the site.
4. After a small amount of time (seems variable) the software changes the set frequency to a different one causing the voice channels to go to the far right edge of the spectrum.
5. The voice channels then get poor decode and fade. They decode perfectly when the manual frequency I set is in use for a brief time.

**Expected behavior**
Respect the manual frequency entered on the tuner tab and don't change it automatically. 

**Screenshots**
N/A

**Application Log**
N/A

**Desktop (optional - complete the following information):**
 - OS: Windows 10 Pro 22H2
 - CPU Cores:  4 cores
 - RAM: 16GB

**Additional context**
As mentioned above, all 6 channels (5 voice, 1 control) are within 7.1 ish MHz spectrum. The tuner is capable of 10 MHz spectrum. The auto generated spectrum is not the middle point between all the channels. It is putting the spectrum skewed to the right causing a couple channels to be at the edge. The system is a P25 phase 1 system (WV SIRN).

DSheirer, 2024-05-19:
The tuner control panel has controls for specifying a minimum and maximum frequency to force the tuner to stay in a frequency range.

ItsTechlogix, 2024-05-20:
Thank you for the response but I'm not seeing any of those type of controls. Here's some screenshots:

[image]

I even found the tuner_confirguration file and tried entering the correct center frequency there but the software still changes to a different one on its own.

[image]

I'm hoping it's just something I am overlooking.

DSheirer, 2024-05-22:
My apologies.  That feature is in the nightly release version and will be in the upcoming beta 2 release for 0.6.1

DSheirer, 2024-05-26:
I just reread your issue.  sdrtrunk manages the tuner's center frequency automatically.  This is a common user misunderstanding.  You do not have to set the center frequency manually.  sdrtrunk does not work like other software applications.

Closing.

closed 2024-05-26 by DSheirer