DSheirer/sdrtrunk #1926: Stops sending calls after hitting error about "toolkit not initialized"

ms4sman, 2024-05-29:
**sdrtrunk Version**
0.6.0

**Describe the bug**
I'm running SDR-Trunk headless on a Raspberry Pi and it's been working great for several months now. But about a month or so ago, it suddenly started encountering some kind of fatal error almost every day that causes it to stop sending calls to my various servers. The Pi reboots every night, so after the reboot, it starts working again for a while, but most days, sometime during the day, it will encounter this error (I'll include below) and stop sending calls. Oddly, this error often appears in the log a couple of times BEFORE it stops sending calls. It's often not until the second or third time the error appears that it stops sending calls. Also, nothing appears in the logs folder about this. This information is from journalctl for the systemd service that I configured for SDR-Trunk.

**To Reproduce**
Unfortunately I am not sure how to tell you to reproduce it, because I'm not sure why it suddenly started happening. I haven't touched the configuration of this pi in months, and it just suddenly started happening.

**Expected behavior**
I'd expect it to not encounter this error, whatever it is, and continue sending calls without stopping.

**Screenshots**
I do not have any screenshots relevant

**Application Log**
`May 28 11:16:34 pi-scanner sdr-trunk[681]: May 28, 2024 11:16:34 AM com.google.common.eventbus.EventBus$LoggingHandler handleException
May 28 11:16:34 pi-scanner sdr-trunk[681]: SEVERE: Exception thrown by subscriber method startChannelRequest(io.github.dsheirer.controller.channel.event.ChannelStartProcessingRequest) on subscriber io.github.dsheirer.controller.channel.ChannelProcessingManager@5bb8f9e2 when dispatching event: io.github.dsheirer.controller.channel.event.ChannelStartProcessingRequest@4d331b25
May 28 11:16:34 pi-scanner sdr-trunk[681]: java.lang.IllegalStateException: Toolkit not initialized
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at javafx.graphics/com.sun.javafx.application.PlatformImpl.runLater(Unknown Source)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at javafx.graphics/com.sun.javafx.application.PlatformImpl.runLater(Unknown Source)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at javafx.graphics/javafx.application.Platform.runLater(Unknown Source)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.controller.channel.ChannelProcessingManager.startProcessing(ChannelProcessingManager.java:393)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.controller.channel.ChannelProcessingManager.startChannelRequest(ChannelProcessingManager.java:318)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(Unknown Source)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at java.base/java.lang.reflect.Method.invoke(Unknown Source)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.Subscriber.invokeSubscriberMethod(Subscriber.java:85)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.Subscriber$SynchronizedSubscriber.invokeSubscriberMethod(Subscriber.java:142)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.Subscriber.lambda$dispatchEvent$0(Subscriber.java:71)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.util.concurrent.DirectExecutor.execute(DirectExecutor.java:31)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.Subscriber.dispatchEvent(Subscriber.java:68)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.Dispatcher$PerThreadQueuedDispatcher.dispatch(Dispatcher.java:109)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at com.google.common.eventbus.EventBus.post(EventBus.java:267)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processPhase1ChannelGrant(P25TrafficChannelManager.java:365)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processChannelGrant(P25TrafficChannelManager.java:252)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processChannelGrant(P25P1DecoderState.java:335)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBKGroupVoiceChannelGrant(P25P1DecoderState.java:1745)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBK(P25P1DecoderState.java:1096)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:310)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:155)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:115)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:108)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:92)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:34)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:327)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:204)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:70)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:87)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.psk.DQPSKGardnerDemodulator.calculateSymbol(DQPSKGardnerDemodulator.java:88)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:146)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:121)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:130)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderLSM.receive(P25P1DecoderLSM.java:39)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:143)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:40)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.filter.channelizer.output.TwoChannelOutputProcessor.process(TwoChannelOutputProcessor.java:124)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.lambda$new$0(ChannelOutputProcessor.java:57)
May 28 11:16:34 pi-scanner sdr-trunk[681]:         at io.github.dsheirer.util.Dispatcher.process(Dispatcher.java:171)
[... cut ...]
**Desktop (optional - complete the following information):**
Raspberry Pi 4
Raspbian
2 GB memory

**Additional context**
I know Raspberry Pi isn't recommended as the ideal platform for this, and 2 GB of memory is also pretty sparse, but I'm only monitoring a single system, using a single SDR stick. I've monitored the memory usage and it appears to not be even close to running out of memory or anything. The system doesn't seem to have any trouble keeping up with the one stick/one system I have it monitoring.

closed 2024-05-30 by DSheirer

ms4sman, 2024-05-30:
@DSheirer thank you for the quick fix on this! I'm downloading the nightly now to test it out and will let you know if I still have any issues.

culleydog2005, 2024-08-01:
Polk County police scanner for all sites down since Saturday and wasn't able to contact you but this way

PR #1928 (2024-05-30): #1926 Toolkit not initialized error when running headless during APCO…
Closes #1926 
Toolkit not initialized error when running headless during APCO25 channel start processing request.