DSheirer/sdrtrunk #1933: SDRPlay API 3.15 support

rlkernsGIT, 2024-06-01:
SDRTrunk v0.6.1-beta

SDRPlay API is delivering API 3.15 now and this is currently not supported by SDRTrunk. Not much different than 3.14. Think they added support for a new radio.

20240531 154410.522 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [18MB/27MB 66%]
20240531 154410.600 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [20MB/27MB 76%]
20240531 154410.616 [main] WARN  i.g.d.s.t.sdrplay.api.Version - Unrecognized SDRplay API version [3.15]  [20MB/27MB 76%]

DSheirer, 2024-06-01:
This is already implemented.  You can use the nightly release until the next version is released

closed 2024-06-01 by DSheirer

icen521, 2024-06-02:
I installed the nightly release from today (June 2, 2024) and unfortunately I'm still seeing the error mentioned above:

2024-06-02 09:17:26.325 INFO  i.g.d.s.t.s.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [19MB/34MB 57%]
2024-06-02 09:17:26.372 INFO  i.g.d.s.t.s.api.SDRplay - API library - open status: Success  [22MB/34MB 64%]
2024-06-02 09:17:26.372 WARN  i.g.d.s.t.s.api.Version - Unrecognized SDRplay API version [3.15]  [22MB/34MB 64%]
2024-06-02 09:17:26.372 INFO  i.g.d.s.t.s.api.SDRplay - API library is not available.  [22MB/34MB 64%]

I have an RSPdx-r2 I'm trying to use SDRTrunk with.

icen521, 2024-06-02:
Nevermind - needed to completely delete the sdr-trunk-windows-x86_64-vnightly folder and re-create from the zip. Simply overwriting the existing folder's contents did not work. Working now.

rlkernsGIT, 2024-06-02:
I just downloaded the nightly build to follow up on the issue. I do not see the 3.15 folder merged/pushed in the source for the nightly build yet. And, I get the same results i.e.

20240602 103437.926 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [18MB/27MB 68%]
20240602 103438.022 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [20MB/27MB 77%]
20240602 103438.034 [main] WARN  i.g.d.s.t.sdrplay.api.Version - Unrecognized SDRplay API version [3.15]  [7MB/27MB 26%]

rlkernsGIT, 2024-06-02:
OK confirmed, It's working fine.