DSheirer/sdrtrunk #1937: P25 trunk calls not showng up in active call window or being recorded

Seagravebuff60, 2024-06-09:
**sdrtrunk Version**
 'master-branch' (nightly 060224) 

**Describe the bug**
P25 trunk call events do not appear in the active call window when a call is in progress. Audio is also not recorded. This happens with both a Harris and Motorola P25 systems. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run SDR Trunk
2. Auto-start all channels 
3. Wait for a P25 trunk call event to happen
4. Notice that it won't show up in the active calls window 
5. Check recordings folder
6. Check TR and notice that the call has not been recorded 

**Expected behavior**
When a call is in progress, it should appear in the events window and be recorded as well. 

**Screenshots**
This group call is active but only shows up in the event window and not the active call window: 
[image]

No recordings in the recordings folder:
[image]

No recordings in Trunking Recorder for that time:
[image]

**Application Log**
[sdrtrunk_app.log](https://github.com/user-attachments/files/15754017/sdrtrunk_app.log)

**Desktop Specs**
 - OS: Windows 10
 - CPU Cores: 6 Cores
 - RAM: 16 GB

Seagravebuff60, 2024-06-10:
I also wanted to add that im seeing the same behavior on another install of SDR Trunk on a separate computer

kyakhta, 2024-06-11:
I see the same behavior with current release and today nightly build on Dell Precision/Win 11 ( i7-13800H/64gb RAM)  
MP3 will show up randomly some time (6-7 files in span of 12 hours on relatively busy site)

Seagravebuff60, 2024-06-30:
> update: after changing LAME Encoding settings from CBR 16 kbps (default) to VBR Better Quality recordings begin working as expected.

Thanks. I tried this, too, but I have an issue with a P25 Motorola System. It seems to be fixed on a Hariis system, though.

DSheirer, 2024-08-24:
@Seagravebuff60 I've tested the latest code base against a Motorola Phase 2 system and everything looks like it's working correctly, against my local test recordings.

That doesn't mean that you're not having an issue.  But, it's impossible for me to diagnose what's going on in your environment if I can't replicate it to test against.

Could you please make a baseband recording that demonstrates the issue where you see that a call is in progress but the traffic channel is not being allocated and decoded? 

Also, please confirm that your control channel configuration for maximum traffic channels is large enough to allow multiple traffic channels to be decoded at the same time.

DSheirer, 2024-10-01:
@Seagravebuff60 this issue may be related/same as #1996 that I just fixed.

For the configuration where this was happening, did you have multiple frequencies listed in the control channel rotation list, and do those frequencies overlap with the traffic channel frequencies?

If so, can you please test against the latest nightly release to see if that clears the issue for you?

Seagravebuff60, 2024-10-01:
Hey Denny, 

Thanks for looking into this. Yes, I have all the frequencies listed in the control channel rotation list, and they do overlap with the traffic channel frequencies because it is a Harris system, and all frequencies can be used as a control channel at any time. 

It works as expected after running it for a few hours. I was running into an issue where the transmission would get cut off at the end, but I changed the encoder settings to CBR, and it is fine now and working as expected. 

Thanks, 
Seagravebuff60

closed 2024-10-02 by DSheirer