DSheirer/sdrtrunk #1948: OpenMHz went down today and SDRTrunk wouldn't start after that..

Ctpolicelive, 2024-08-12:
**sdrtrunk Version**
any with Openmhz feature

**Describe the bug**
OPENMhz went down today for whatever reason and it took my other broadcastify feeds with it. Sdrtrunk wouldn't even restart. When restarting my it kept failing on "Connecting to OpenMhz feeds.."

I had to manually edit my playlist XML file with a XML editor, and change the Openmhz server startup to false.

**Desktop (optional - complete the following information):**
-Linux Mint

**Additional context**
Need some kind of option to retry openmhz server 3 times at start up and then continue without it, if it fails.

Ctpolicelive, 2024-08-14:
Happened again today. I have 5 Openmhz and 8 broadcastify streams. When Openmhz goes down it forces sdrtrunk to disconnect my other broadcastify streams and no other streams will restart until either openmhz is responsive, or I disable my openmhz streams. 

I can see sdrtrunk tries to reconnect but can't get past that many down openmhz servers in time to reconnect to broadcastify servers. It gets stuck in a loop.

So a simple tweak may be needed in the reconnect timing of streams so it doesn't take down everything if openmhz goes down.

DSheirer, 2024-08-14:
Can you please post the application log from when this happens so I can see if there are any logged errors.

Ctpolicelive, 2024-08-14:
[sdrtrunk_app.log](https://github.com/user-attachments/files/16617706/sdrtrunk_app.log)

20240814 085101.944 [sdrtrunk channel [113/P25-2] 852262500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 51%]
20240814 094206.392 [sdrtrunk channel [90/P25-2] 852262500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 52%]
20240814 094513.008 [sdrtrunk channel [106/P25-2] 852262500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 53%]
20240814 104254.631 [ForkJoinPool.commonPool-worker-104] ERROR i.g.d.a.b.b.BroadcastifyCallBroadcaster - Broadcastify calls API file upload fail [503] response [<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>SlowDown</Code><Message>Please reduce your request rate.</Message><RequestId>T1PGXQ2X9D2209D5</RequestId><HostId>6R6oxhdxhAMJDwyY66/WEiNxuiosW+HmE7b67Ncc4RmgS/JJ4Hg5gkpiEqLTySF/RCITmg5q4Iw=</HostId></Error>]  [899MB/2GB 38%]
20240814 123911.991 [sdrtrunk channel [103/P25-2] 852262500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 67%]
20240814 124940.012 [sdrtrunk channel [90/P25-2] 853387500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 56%]
20240814 125219.243 [sdrtrunk channel [94/P25-2] 853387500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [1GB/2GB 84%]
20240814 130442.944 [sdrtrunk channel [101/P25-2] 853387500 thread 1] DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [2GB/2GB 89%]
20240814 132239.337 [sdrtrunk scheduled thread 1] ERROR i.g.d.a.b.BroadcastModel - Error deleting temporary internet recording file: /home/useme/SDRTrunk/streaming/temporary_streaming_file_10050_20240814_132221.276.mp3 - /home/useme/SDRTrunk/streaming/temporary_streaming_file_10050_20240814_132221.276.mp3  [1GB/2GB 72%]
20240814 133018.022 [sdrtrunk scheduled thread 3] INFO  i.g.d.s.t.manager.TunerManager - Tuner plug-in detected at USB Bus [3] Port [10] Tuner Class [RTL-2832]  [1GB/2GB 77%]
20240814 133018.023 [sdrtrunk scheduled thread 3] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:3 Port:10 - Added / Starting ...  [1GB/2GB 77%]
20240814 133019.033 [sdrtrunk scheduled thread 3] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [1GB/2GB 60%]
20240814 133131.076 [AWT-EventQueue-0] INFO  i.g.d.s.tuner.ui.TunerEditor - Disabling RTL-2832 tuner  [1GB/2GB 60%]
20240814 133131.077 [AWT-EventQueue-0] INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:3 Port:7  [1GB/2GB 60%]
20240814 133137.116 [AWT-EventQueue-0] INFO  i.g.d.s.tuner.ui.TunerEditor - Disabling RTL-2832 tuner  [1GB/2GB 45%]
20240814 133137.116 [AWT-EventQueue-0] INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:3 Port:10  [1GB/2GB 45%]
20240814 133154.953 [AWT-EventQueue-0] INFO  i.g.d.s.tuner.ui.TunerEditor - Enabling RTL-2832 tuner  [1GB/2GB 59%]
20240814 133155.941 [AWT-EventQueue-0] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [1GB/2GB 63%]
20240814 133203.928 [AWT-EventQueue-0] INFO  i.g.d.s.tuner.ui.TunerEditor - Enabling RTL-2832 tuner  [1GB/2GB 85%]
20240814 133204.912 [AWT-EventQueue-0] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [1GB/2GB 67%]
20240814 133207.705 [sdrtrunk channel [5/P25-1] 853912500 thread 1] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [1GB/2GB 64%]
20240814 133248.835 [sdrtrunk scheduled thread 3] INFO  i.g.d.s.t.u.DiscoveredTunerModel - Removing discovered tuner: RTL-2832 USB Bus:3 Port:10  [1GB/2GB 80%]
20240814 133248.836 [sdrtrunk scheduled thread 3] INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:3 Port:10  [1GB/2GB 80%]
20240814 133248.882 [sdrtrunk scheduled thread 3] INFO  i.g.d.s.t.manager.TunerManager - Tuner Unplugged: RTL-2832 USB Bus:3 Port:10  [1GB/2GB 83%]
20240814 141339.332 [sdrtrunk scheduled thread 1] ERROR i.g.d.a.b.BroadcastModel - Error deleting temporary internet recording file: /home/useme/SDRTrunk/streaming/temporary_streaming_file_10778_20240814_141319.276.mp3 - /home/useme/SDRTrunk/streaming/temporary_streaming_file_10778_20240814_141319.276.mp3  [1GB/2GB 55%]
20240814 144719.625 [ForkJoinPool.commonPool-worker-105] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - OpenMHz API file upload fail [504] response [<html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.27.0</center>
</body>
</html>
]  [1GB/2GB 46%]
20240814 144719.895 [ForkJoinPool.commonPool-worker-106] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - OpenMHz API file upload fail [504] response [<html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.27.0</center>
</body>
</html>
]  [1GB/2GB 57%]
20240814 144719.895 [ForkJoinPool.commonPool-worker-105] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - OpenMHz API file upload fail [504] response [<html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
[... cut ...]
20240814 145126.952 [sdrtrunk scheduled thread 4] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - Exception connecting to OpenMHz server [java.lang.InterruptedException]  [910MB/2GB 39%]
20240814 145126.952 [sdrtrunk scheduled thread 4] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - Error reconnecting to OpenMHz server [Uknown Exception]  [910MB/2GB 39%]
20240814 145136.032 [sdrtrunk scheduled thread 2] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - Exception connecting to OpenMHz server [java.lang.InterruptedException]  [1GB/2GB 76%]
20240814 145136.032 [sdrtrunk scheduled thread 2] ERROR i.g.d.a.b.o.OpenMHzBroadcaster - Error reconnecting to OpenMHz server [Uknown Exception]  [1GB/2GB 76%]

Ctpolicelive, 2024-08-19:
Had to shut down my Openmhz streams until this can be fixed. It keeps happening. Some kind of skew issues apparently. If Openmhz website goes down for even one second it takes all my other feeds with it. Not sure if Openmhz is being DDOSed or just overwhelmed. Either way it shouldn't be tied at the hip to my other streams. Thanks for taking a look at this.

tadscottsmith, 2024-08-20:
Are you on at least v0.6.1-beta-1? There were some OpenMHz reconnection issues addressed in that version.

Ctpolicelive, 2024-08-26:
> Are you on at least v0.6.1-beta-1? There were some OpenMHz reconnection issues addressed in that version.

Yes, on a recent nightly. Doesn't matter which version. Happen on all. Just happened again today. Openmhz goes down and my other feeds won't reconnect until I disable Openmhz. The log I provided above seems to explain whats happening.  Thanks

DSheirer, 2024-09-03:
OpenMHz is configured to take up to 20 seconds trying to connect before it fails with a timeout.  This was happening on the main application thread and may have been what was causing the perception that the application was failing to start.

I've updated the code to spin off all stream configuration startups into their own threaded startup so that they don't cause any delay in the main application start sequence.

I'm not sure what was causing the Broadcastify Calls streaming delays with the call time skew.  From the posted log file, it appears that OpenMHz was at least partially responding from the nginx gateway timeout responses.

closed 2024-09-03 by DSheirer

Ctpolicelive, 2024-09-06:
Still happening in latest nightly.. Openmhz goes down and sdrtrunk takes everything down with it,. 
[sdrtrunk_app(6).log](https://github.com/user-attachments/files/16912314/sdrtrunk_app.6.log)

PR #1968 (2024-09-03): #1948 Audio Streaming Startups On Separate Thread
Closes #1948 

Audio Stream starts are now spun off to a cached thread pool to prevent any initial stream connections from delaying the overall application startup.