DSheirer/sdrtrunk #1970: Stream audio glitches out after about a day of running on latest nightly

Ctpolicelive, 2024-09-05:
**sdrtrunk Version**
Only Last 2 or 3 nightly's have this issue. Running much earlier nightlys, or final, does NOT do this.

**Describe the bug**
After about a day or two of running ALL audio gets a nasty digital glitching sound across all streams. 

**To Reproduce**
Run latest nightly for a day or two and my scanner streams all get a digital glitching sound.

**Expected behavior**
All scanner stream's audio starts getting a horrible noisy digital glitch sound with every call. 

**Application Log**
Uploaded log. Note that I switched back to an older working nightly in this log. I also turned off the preferred tuner error which was filling up the logs and unsure if related to my issue. 

**Desktop (optional - complete the following information):**
 - OS: LINUX
 - CPU Cores: 28
 - RAM: 16

**Additional context**

[sdrtrunk_app(5).log](https://github.com/user-attachments/files/16883439/sdrtrunk_app.5.log)

DSheirer, 2024-09-05:
Unfortunately, I don't see anything in the application log that would explain the issues that you're seeing.

There is another (unrelated) repeating error that I see in your app log - I created a new issue for that.

DSheirer, 2024-09-05:
Does the 'nasty digital glitching' sound only happen on the local PC audio playback or is it also on the streamed audio?  If you open the MP3 audio recordings and use a separate playback application, does the audio have this same characteristics?

Ctpolicelive, 2024-09-06:
I'm not sure if its on local playback also, but comes out on all streams. Other things I've also noticed with these new nightlys is I see the green data will flash on my control channel a lot. I have data calls ignored. Also I seeing constant teardowns which doesn't happen on previous nightlys. And the events log disappears, like it keeps clearing itself,.

ki5zha, 2024-09-08:
Been having the same issues with my setup lately on the latest build.
[20240907_sdrtrunk_app.log](https://github.com/user-attachments/files/16923410/20240907_sdrtrunk_app.log)

I use 3 RTL_SDR dongles (reduced now to 2 with one disabled).  and have audio processing errors.  The tones get jittery.

Didnt have this problem with the later versions.

Ctpolicelive, 2024-09-10:
Latest nightly does the same thing. Something is glitchy in these new nightlys. As far as I can tell the digital screech sound doesn't happen on the local mp3s when i listen to the temp stream files. I listened to a bunch of files when it started to happen and could hear it. I could be wrong on that though. Very odd. It's like clockwork, after a day or so every call gets a digital sound. And this isn't the everyday interference digital garble we all know to love.. Been running sdrtrunk for 5 years, It's only on the latest nightlys this issue happen. I will provide the latest log maybe you can find something in there. I do see some new usb errors.. 

Thanks

[sdrtrunk_app(7).log](https://github.com/user-attachments/files/16939200/sdrtrunk_app.7.log)

closed 2024-09-17 by Ctpolicelive