DSheirer/sdrtrunk #1973: Harris P25 Patch/Supergroup Behavior

Skyking-Bravo, 2024-09-08:
**sdrtrunk Version**
Nightly Version (Downloaded on 8/30/24)

**Describe the bug**
I am using sdrtrunk to stream to Rdio-Scanner and on the Harris-built Miami P25 system [https://www.radioreference.com/db/sid/10812](https://www.radioreference.com/db/sid/10812) many transmission only show up on the patch 65XXX talkgroups and not also the source talkgroups such as Fire Dispatch (TG 1801) and EMS Dispatch (1803). It seems like it should be possible in sdrtrunk to generate the audio files in such a way that Rdio-Scanner would play the traffic on the Fire or EMS Dispatch channels. When the 65XXX talkgroups are active the traffic is also heard on the source talkgroups when using my SDS100.

**To Reproduce**
Not Applicable

**Expected behavior**
When a 65XXX series patch talkgroup is in use, I expect sdrtrunk to produce a streaming file that contains the necessary metadata to allow rdio-scanner to play the audio via the Fire-Dispatch or EMS Dispatch channel. 

**Screenshots**
Not Applicable

**Application Log**

**Desktop (optional - complete the following information):**
 - OS: Ubuntu 24.04
 - CPU Cores: 12
 - RAM: 16Gb

**Additional context**
Examples of filenames of audio recordings:
'20240908_004337Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030220.wav'
'20240908_004348Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030901.wav'
'20240908_004355Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030275.wav'
'20240908_004401Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030901.wav'
'20240908_004432Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030273.wav'
'20240908_004448Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030901.wav'
 20240908_004505Miami_P25_Miami_T-Control__TO_65144_FROM_7030198.wav

Details of the first file:
File Name                       : 20240908_004337Miami_P25_Miami_T-Control__TO_P_65144_[8390409__8390411]_FROM_7030220.wav
Directory                       : .
File Size                       : 47 kB
File Modification Date/Time     : 2024:09:08 00:43:37-04:00
File Access Date/Time           : 2024:09:08 00:44:31-04:00
File Inode Change Date/Time     : 2024:09:08 00:43:37-04:00
File Permissions                : -rw-rw-r--
File Type                       : WAV
File Type Extension             : wav
MIME Type                       : audio/x-wav
Encoding                        : Microsoft PCM
Num Channels                    : 1
Sample Rate                     : 8000
Avg Bytes Per Sec               : 16000
Bits Per Sample                 : 16
Product                         : T-Control
Subject                         : Miami P25
Warning                         : Bad CMT� chunk
ID3 Size                        : 393
Comment                         : Date:2024-09-08 00:43:37.946;System:Miami P25;Site:Miami;Name:T-Control;Decoder:P25 Phase 1;Channel:0-756;Frequency:860462500;
Album                           : T-Control
Composer                        : sdrtrunk nightly - 2024-08-31T12:34:00.580+0000
Genre                           : Scanner Audio
Date                            : 2024-09-08 00:43:37.946
Grouping                        : Miami P25
Title                           : P:65144 [8390409, 8390411]"Patch Channels"
Artist                          : 7030220
Year                            : 2024
Date/Time Original              : 2024
Duration                        : 2.93 s

DSheirer, 2024-09-09:
What setting are you using for patch group streaming?

Go to the User Preferences dialog, Audio >> Call Management tab.

You can select the behavior for patch group streaming to either stream the call audio using the patch group, or to stream the call as the individual patched groups.

Skyking-Bravo, 2024-09-09:
> What setting are you using for patch group streaming?
> 
> Go to the User Preferences dialog, Audio >> Call Management tab.
> 
> You can select the behavior for patch group streaming to either stream the call audio using the patch group, or to stream the call as the individual patched groups.

Yes, I am familiar with those options. Selecting "Individual Talkgroups" does not help. In fact it seems to make things worse in that I appear to be missing some traffic entirely, when compared to selecting "Patch Group".

Skyking-Bravo, 2024-09-12:
Maybe these screenshot might be helpful. In the first, user preference for streaming is set to individual talkgroups and in the second it is set to patch group. In the first screen shot, Fire Dispatch 1801 is patched with 65133 according to my SDS100 and with 65130 in the second screenshot. But it doesn't look like sdrtrunk is seeing 1801 is either scenario.
[image]
[image]

DSheirer, 2024-10-06:
@Skyking-Bravo sorry for the delay.  Rdio-Scanner works with sdrtrunk in 2x ways:

1. Rdio-Scanner monitors sdrtrunk's recording directory and selects/streams calls from the recordings
2. sdrtrunk sends the calls to Rdio-Scanner's API

Which option are you using?

boyd-glover, 2024-10-10:
> @Skyking-Bravo sorry for the delay. Rdio-Scanner works with sdrtrunk in 2x ways:
> 
> 1. Rdio-Scanner monitors sdrtrunk's recording directory and selects/streams calls from the recordings
> 2. sdrtrunk sends the calls to Rdio-Scanner's API
> 
> Which option are you using?

I am using the API. I have looked at both of the streaming files and the recorded files and in both cases there’s only a file for the patched talkgroup in the 6XXXXX-series.

DSheirer, 2024-10-14:
@Skyking-Bravo I created some automated tests to verify that the patch group streaming works correctly according to the setting in the User Preferences for Call Management to verify that the code is working correctly.

The only thing I can't confirm is if sdrtrunk is seeing the patch groups the same way that your SDS is seeing them.  You can view the 'Details' tab to list the patch groups that are active, hitting the refresh button repeatedly.  You might compare if sdrtrunk is showing the same patch groups as active on the system as your SDS.  That could impact the streaming of patch groups if sdrtrunk is not tracking the same as the SDS.

DSheirer, 2024-10-14:
Also, when you're looking at recordings for streaming, you'll need to look in the 'streaming' folder to see the temporary recording files that are created for streaming.  However, once the recording gets streamed, it is deleted immediately and that could make it a challenge to see what's being generated.

The audio recordings that you listed above that you'll find in the 'recordings' folder are the recordings generated by your aliases that are set to record.  If any of the aliases, such as the patch group identifier or any of the patched talkgroup identifiers, are set to record, then the call is recorded, just once.

Seagravebuff60, 2024-10-19:
Ive noticed this as an issue, too. However, I am using trunking recorder. I have noticed that when a Harris Simulselect patch is granted, the Subgroup and Supergroup in the SDR Trunk events window and on the detalias tab are wrong. Further leading Trunking Recorder to show inaccurate info. 

In this example, Talkgroup 65190 is the Supergroup. SDR Trunk shows that "8388722, 8388717, 8388818" are the subgroups in the patch. When Talkgroups 114, 109, and 210 are really the subgroups.  I can confirm this with Unitrunker (see the 2nd screenshot)
[image]
[image]

Moreover, here is another (different) example. SDR Trunk is telling Trunking Recorder that TGs 8388721 and 8388716 are the subgroup of TG 65151. In reality, TGs 108 and 113 are the subgroups of supergroup 65151, according to Unitrunker. (see the 2nd screenshot)
[image]
[image]

Again, this is and has been an SDR Trunk Issue, not a Unitrunker or Trunking Recorder issue. It also relates to my previous GitHub bug report. I wonder why it was closed when it wasn't fully fixed. 
https://github.com/DSheirer/sdrtrunk/issues/1607

DSheirer, 2024-10-19:
@Seagravebuff60 I created a new issue to fix the Harris patched talkgroup issue here: https://github.com/DSheirer/sdrtrunk/issues/2035

The original issue #1607 was to add Patching support for both Moto and Harris for both Phase 1 and 2, because that support didn't exist at all previously.  That feature was delivered and merged into the master branch and that's why the issue number is closed.  Once an issue is merged into master branch, I generally don't reopen the issue and any followup work is handled under a new issue number.

DSheirer, 2024-10-19:
@Skyking-Bravo the fix that I'm applying under issue #2035 may resolve the issue you're reporting under this #1973 issue number.  But, I'm not fully clear from your description, if the fix that I'm applying in #2035 fully fixes your issue.  So, please test against the nightly once I merge #2035 and let me know if we can close this issue, or if there is further needed work to address streaming/metadata.

Skyking-Bravo, 2024-10-19:
> @Skyking-Bravo the fix that I'm applying under issue #2035 may resolve the issue you're reporting under this #1973 issue number. But, I'm not fully clear from your description, if the fix that I'm [...]

I downloaded the nightly build this morning and have been testing it for the past few hours. It is now working as expected. Thanks so much for addressing the issue.

closed 2024-10-19 by Skyking-Bravo