DSheirer/sdrtrunk #1988: SDR Trunk Nightly - P25 Trunking Missed calls

Tbolt10033, 2024-09-20:
**sdrtrunk Version**
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.
SDR-Trunk Nightly - Latest Version as of 9/20/24

**Describe the bug**
Sometimes, I miss a call even though I should be able to hear all calls like with version 0.6.0. Occasionally, I only hear the second transmission in a conversation, missing the first one. For example, I might hear "Roger pump xx back at quarters" but not "Dispatch Pump xx back at quarters." This issue started after some changes to P25 decoder were made, I believe this: https://github.com/DSheirer/sdrtrunk/pull/1898

**To Reproduce**
Unknown

**Expected behavior**
Like version 0.6.0 the latest version should be able to hear all calls and not miss a call.

**Radio System Protocol**
P25 Trunking Phase I CC with Phase II TDMA talkgroups

DSheirer, 2024-10-14:
@Tbolt10033 I've made a lot of changes to the codebase.  Please test again with the nightly release and let me know if you're still seeing missged calls?

Tbolt10033, 2024-10-14:
> @Tbolt10033 I've made a lot of changes to the codebase. Please test again with the nightly release and let me know if you're still seeing missged calls?

I think it is working now. This can be closed.

closed 2024-10-14 by DSheirer