DSheirer/sdrtrunk #1989: P25 Phase 1 Traffic Channels Shutdown Prematurely

DSheirer, 2024-09-21:
**sdrtrunk Version**
master - nightly release.

**Describe the bug**
As originally reported by @rxes in #1915

P25 Phase 1 calls do not playback audio.  It appears the issue is the traffic channels are shutting down prematurely before producing any audio.  In both sets of recordings, the traffic channel bitstream recording files appear to indicate that the traffic channel is being torn down after less than 1-2 seconds of decode.

This would normally be caused by a poor decode quality and the channel timer kicks in and shuts down the channel.  However, the recording samples indicate good decodes that should have triggered the fade timer update look and allowed the channel to continue to decode.  Not sure whats causing the premature teardowns.

[recordings.zip](https://github.com/user-attachments/files/17084234/recordings.zip)

[recordings(1).zip](https://github.com/user-attachments/files/17084235/recordings.1.zip)

rxes, 2024-09-21:
The last working nightly, in my case, was from 2-13-24. The changes made on 4-26-24 broke something and since then no nightly has worked properly. I hope that helps

DSheirer, 2024-09-23:
@rxes I added some 'forensic' logging to the traffic channels to better see what is going on and causing the premature traffic channel teardowns.  This version will create text files in the recording directory for each call/traffic channel.

Can you please run the software through 5 or 6 call events and then zip up the .bits recordings and the forensic log files and attach to this issue?

The download link for my google drive above is still valid for the latest sdrtrunk builds: 

https://drive.google.com/drive/folders/130eCl0Bv-G7RlsBMtIfiP9ZhdCmg_FpL?usp=sharing

rxes, 2024-09-23:
[recordings 9-23-24.zip](https://github.com/user-attachments/files/17104316/recordings.9-23-24.zip)

rxes, 2024-09-30:
I'm not sure if the newest nightly was supposed to fix the bug in this post, but it didn't change anything.
[9-29-24 recordings.zip](https://github.com/user-attachments/files/17182476/9-29-24.recordings.zip)

DSheirer, 2024-09-30:
The latest code change was to fix something else ... subsequent calls on a traffic channel were sometimes not detecting the encryption type correctly.

rxes, 2024-10-01:
For now it appears that latest nightly fixed the problem I was experiencing. Thank you

DSheirer, 2024-10-02:
Thanks!  I'll close this issue for now, but we can reopen if you see it reappear.

closed 2024-10-02 by DSheirer