DSheirer/sdrtrunk #2008: Add CTCSS codes to Conventional

slowbalt, 2024-10-06:
Please add CTCSS codes, this would make the program just perfect. Thanks!

slowbalt, 2024-10-12:
I'll admit not having any idea what is involved, or how complicated that is. But for me, it is the one feature that would make the program perfect. I have seen the idea of "bounties". How much would be appropriate? I'd pitch in 50$ in a heart beat and I can hardly believe I'm alone!

grahampaull, 2024-10-15:
This would be a very welcome addition.

hunterdave1979, 2024-10-16:
C5o

seth4380, 2024-10-28:
This would be a perfect addition. Also to get rid of the squelch noise at the end of every transmission. Maybe with adding pl tones that would fix that too.

kernworks, 2025-01-03:
+1 for me as well.

There was discussion from @DSheirer a while ago in #80 

https://github.com/DSheirer/sdrtrunk/issues/80#issuecomment-1585558699

This would be the prefect implementation. Being able to add it as an Identifier to an alias would be amazing and would open doors to allow conventional to be submitted to Broadcastify Calls

Bonus points for adding C**D**CSS as well.

someengineer1, 2025-01-12:
Tossing my hat/vote into the ring too, CTCSS would be great.  Is there still a plan to implement it as an alias like in that other thread?

Been toying with Trunk Recorder on Linux just to get this feature but that's a whole other can of worms.

someengineer1, 2025-03-22:
> Hope DSheirer finds some time to add support for CTCSS/DCS codes, because that is the only thing this program is missing (leave aside TETRA) to be absolutely perfect :)

DCS is already in there, but not CTCSS.  I ended up switching to Trunk Recorder as it can do all my systems and includes CTCSS, but definitely a big undertaking to get it working.  The P25 decoding in SDRTrunk with JMBE decoder is superior by a bit, but I wanted an all-in-one solution.  But may revisit in the future and add SDRTrunk back into the mix if CTCSS gets added (if only it had SmartNet support, then it could be my all in one solution).

pauldonahue, 2025-04-24:
+1 to the request to add CTCSS / PL tone decoding for NBFM

cherizon, 2025-09-28:
+1 to add CTCSS/PL

msullivan1993, 2026-03-13:
> Just spent days implementing sdrtrunk as a backend before realizing it didnt support CTCSS. This is crazy to me, all it takes is a basic tone decoder - I thought all scanners supported CTCSS and DCS at this point.

I thought the same thing, and it was a huge issue for my deployment because of P25 and NXDN co-channels from distant repeaters.

FWIW I have a CTCSS decoder awaiting PR, thanks to Claude. I have been using it on my system and tweaking it as I go along. It’s not perfect but it fixes 90-95% of the issues I was having.

seth4380, 2026-03-15:
> I thought the same thing, and it was a huge issue for my deployment because of P25 and NXDN co-channels from distant repeaters.
>
> FWIW I have a CTCSS decoder awaiting PR, thanks to Claude. I have [...]

How would I get my hands on CTCSS decoder from Claude?

closed 2026-09-26 by DSheirer