DSheirer/sdrtrunk #2009: Add docs how to build from source

D3vil0p3r, 2024-10-08:
Can you please add in docs how to build from source by specifying build dependencies and runtime dependencies separately?

tadscottsmith, 2024-10-08:
https://github.com/DSheirer/sdrtrunk/wiki/Developing

DSheirer, 2024-10-13:
@D3vil0p3r does the link that tadscottsmitth provide the documentation that you're looking for?

D3vil0p3r, 2024-10-13:
Thank you for your answers. I read that the only dependency seems to be OpenJDK 20.x (a non-LTS). Is there no a OpenJDK LTS version compatible?

I'm looking it for Arch Linux. So I guess that the only runtime dependency should be [OpenJDK 20](https://openjdk.org/projects/jdk/20/) and the only build dependency is [gradle](https://archlinux.org/packages/extra/any/gradle/)?

DSheirer, 2024-10-14:
The latest OpenJDK LTS is version 21.  sdrtrunk now requires version 23.

sdrtrunk uses the Java Foreign Function and Memory API for the SDRPlay tuner interface and FFM is still in preview.  

It's possible that I'll land on the next LTS version with OpenJDK 25 in one year from now.