DSheirer/sdrtrunk #2016: Todays nightly won't start.

TravisSamren, 2024-10-14:
**sdrtrunk Version**
Today's nightly

**Describe the bug**
Sdrtrunk won't start

**Application Log**
Gives this error when trying to start.. 

Exception in thread "main" java.lang.NullPointerException: Cannot invoke "io.github.dsheirer.preference.playback.PlaybackPreference.getMixerChannelConfiguration()" because the return value of "io.github.dsheirer.preference.UserPreferences.getPlaybackPreference()" is null
	at io.github.dsheirer.audio.playback.AudioPlaybackManager.<init>(AudioPlaybackManager.java:79)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:218)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:937)

**Desktop (optional - complete the following information):**
 - OS: Linux
 - CPU Cores: 28
 - RAM:32gb

DSheirer, 2024-10-14:
@TravisSamren  Thanks for the bug report and sorry for the bad build.

closed 2024-10-14 by DSheirer

TravisSamren, 2024-10-14:
Glad I could help. Newest build booted up with no errors. All systems nominal. Thanks for the timely fix. I will keep testing.

PR #2017 (2024-10-14): #2016 Playback Preferences Null Error
Closes #2016 Fixes code regression introduced in #1973 with null playback preferences.