DSheirer/sdrtrunk #2041: SDRTrunk nightly crashing

bushcraftbuddy, 2024-10-25:
I have a remote Virtualized SDRTrunk System That will crash if I am not remoted (RDP) into it

I am using 3 RTL-SDR's To Record one P25 Phase 1 System and an NBFM Frequency all of the recordings are then going to Trunking-Recorder for Parsing

when I know it's crashed, I remote in, and it appears to then close the application window. after I reopen the application, it works fine unless I disconnect from the RDP session 

the VM's Resources:

[image]

Nothing appears it happen in the logs. the last thing that shows is the Channelizer Sarting as an INFO log

The last change I made to the System was upgrading from 0.6.0 (final) to the nightly build (I am aware of the bugs, just didn't think this would be one

bushcraftbuddy, 2024-10-28:
Update: I'm now seeing the attached screenshot after a couple hours of running (while being constantly connected via RDP) it stops all call's and doesn't record any of the P25 Channels but does still record the NBFM channel. 

[image]

It still appears to be working based off of the spectrum but nothing else is happening (it's locked on to four channels when only two are actually "doing" anything)

[image]

If you need any more details let me know and I'll get them to you as soon as possible

DSheirer, 2024-10-29:
@bushcraftbuddy I created a separate issue for the 2x error stack traces that you posted.  We're you accessing the message filters at some point?  I'm not ruling these out as contributing to the issue you're seeing, but they might be coincidental.

bushcraftbuddy, 2024-10-29:
I have not done anything with the message filters Since this issue appeared. And that was two weeks ago when I switched to nightly. After it crashes I usually check the log file and then open the application back up and then leave it to auto start the channels. 

This may be a separate issue since the application doesn't just close out immediately but I'm not sure