DSheirer/sdrtrunk #2054: P25 conventional channels missing transmissions.

seth4380, 2024-10-28:
I’m missing a bunch of transmissions when using add-trunk to monitor p25 conventional transmissions. Almost like half are missing and don’t come through. I could never get this to work good with them. My scanner at home picks them up as well as my xts1500 and trunk-recorder works fine. This software is awesome but it would make it better if I could get that to work correctly. Thanks Denny for all the hard work.

DSheirer, 2024-10-28:
@seth4380 can you please make one or more baseband recordings of P25 conventional transmissions that I can use to test against and send them to me or attach them to this issue?

In the channel configuration editor in the recording section, you can turn on 'Baseband I/Q (.wav)" recording option.

The file(s) will be in the SDRtrunk user directory under the recordings folder.

seth4380, 2024-11-03:
Denny, It won't let me upload them here. Do you have another way I can send them to you?

DSheirer, 2024-11-03:
Try this file sharing site: https://upload.disroot.org/#!

kazzaw, 2024-11-14:
Just wondering if there was any progress or updates on this one? I've been having issues with P25 conventional missing transmissions as well, had to go back to tuning into the analog channel instead. Happy to provide baseband recordings as well if required Denny

seth4380, 2024-11-14:
I totally forgot to upload them to Denny. I’ll do that tonight here. Sent from my iPhone

DSheirer, 2024-11-15:
@kazzaw yes please, if you're able to share some recordings.

kazzaw, 2024-11-15:
Thanks[, I've attached one](https://upload.disroot.org/r/UDMExEkn#dEYvnp0hTMt9dt9CheFsNBJMFb8fyhKxMpX7b3PQDMc=). It seemed to capture better than last time, however still seems quite poor. Not too sure if its a reception issue (it shouldn't be, as the tower is only a couple of km away as the crow flies). I might play around with my Airspy and see if I can get some higher quality recordings if thats the case

DSheirer, 2024-11-16:
> Thanks[, I've attached one](https://upload.disroot.org/r/UDMExEkn#dEYvnp0hTMt9dt9CheFsNBJMFb8fyhKxMpX7b3PQDMc=). It seemed to capture better than last time, however still seems quite poor. Not too [...]

That looks like a nice strong signal.  I'm working on it.

someengineer1, 2024-12-09:
> That looks like a nice strong signal. I'm working on it.

I've noticed the same with 0.6.0 and 0.6.1.  Local police converted to P25 Phase 1, untrunked, just a single frequency with no control channel.  Is this potentially something to do with the problem, that the software is more designed for trunked systems (obviously hence the name)?  I notice under messages it logs several "synch loss" every second that there is no transmission, but obviously with this type of system there won't be synch in between calls.  Maybe handling all those synch loss messages is what's overloading the program or something?

About 50% of the transmissions come through nice and clear and the decoding is spot on and high quality, the other 50% are totally skipped or are partial or garbled.  I'm using DSD+ fastlane right now and it catches and decodes everything, but their decoder/vocoder/whatever leaves a lot to be desired in the sound quality department.

The signal is plenty strong and not far away, and the PC is running like 3 to 5% CPU, has plenty of horsepower and memory.  There are several antennas/repeaters around the city and I saw someone mention that they thought it was some sort of simulcast but there is one very close to me that is plenty strong and the others are quite a bit further away.

I have played around with various settings trying to see if I could find something but haven't had any luck.

If you need more samples to troubleshoot I can try to capture some.

someengineer1, 2024-12-11:
Interesting, switching to LSM may have helped (figured I'd give it a try since it is supposedly simulcast, even though it is Phase 1 untrunked).  But will need more time to see if it is just coincidence or actually improved things.

someengineer1, 2024-12-11:
LSM seems to have done it.  If anyone else is watching this with similar issue, give LSM a try even if your system isn't trunked.

GlowingUraniumBerry, 2024-12-13:
I created an account just to confirm LSM fixed my issue. I'm monitoring a single frequency P25p1 system, local PD, and noticed atleast the first 3 or 4 seconds were missing. Switched from C4FM to LSM and voila, problem solved. Thank you for all your great work!

Zachary3352, 2024-12-15:
I've been watching this thread because I'm having the same issue with a P25p1 local PD system as well. @GlowingUraniumBerry I tried switching to LSM, and now SDRTrunk doesn't recognize when calls come in at all. Switching back to C4FM means calls are recognized again, but often audio only starts a few seconds in. Any thoughts?

kazzaw, 2024-12-16:
Yeah, I also changed to LSM with no success. As per above, switching back to C4FM calls are once again recognized, however are scratchy. 

I think we need to wait and see what Denny comes back with

someengineer1, 2024-12-16:
Potentially signal issues?  Bump up the gain or improve the antennas and see if it helps?

seth4380, 2024-12-16:
Switching to lsm didn’t work for me in the past. Sent from my iPhone

Zachary3352, 2024-12-17:
@someengineer1 Theoretically possible, but I doubt it. I've used this SDR in this location for about two years with the SDR# Simple APCO plugin and I've never missed a message. The SDR is an NESDR Smart v4 with the kit UHF antenna. In sdrtrunk, gain is set to 420, sample rate 2.4 MHz. I also have a Motorola XTS and an XTL that both receive fine (both unaffiliated).

In C4FM, once messages start coming in, they sound fine. In LSM mode, I see the signal on the scope clearly, but sdrtrunk doesn't recognize the transmissions.

someengineer1, 2024-12-17:
> @someengineer1 Theoretically possible, but I doubt it. I've used this SDR in this location for about two years with the SDR# Simple APCO plugin and I've never missed a message. The SDR is an NESDR Smart v4 with the kit UHF antenna. In sdrtrunk, gain is set to 420, sample rate 2.4 MHz. I also have a Motorola XTS and an XTL that receive fine (both unaffiliated).
> 
> In C4FM, once messages start coming in, they sound fine. In LSM mode, I see the signal on the scope clearly, but sdrtrunk doesn't recognize the transmissions.

42db is very high gain.  Also if you're only listening to one frequency on the dongle, drop it down to the lowest bandwidth, like 230khz I think?  Significantly reduces the processing load and lets it focus on a much narrower range.  Even if you have two frequencies close together, drop it down to just above what you need to cover both.

Even though you can't hear it, digital still has the static burst at the beginning of the transmission.  Maybe JMBE is more sensitive to that than the other decoders, or maybe your gain is higher in SDRTrunk than it was in SDR#?  Setting the gain too high jacks up your noise floor, and to "latch on" to the signal you need a decent space between noise and signal (good SNR). 

Look at it this way, build a solid foundation before trying to build a house.  The antenna that comes with the Nooelec is a monopole, not even sure if it has a decent ground plane.  For P25, a half wave dipole works much better.  You can build a good one out of 12 or 10 gauge electrical wire cut to the right lengths.  Mine are up in the attic (height makes all the difference, but if your signals are weak, you might even need to get one outside).  I'm running Blog v4 dongles at 25.x db gain and my noise floor is -60 and signal is -10.  Technically that's a bit "hot" (-20 is probably where you normally want to target) but it is providing excellent decodes, and on the analog channel I listen to it helps the squelch attack quicker and reduces the static, my area still uses CTCSS squelch and SDRTrunk doesn't support that, so I have to deal with it other ways.

Look on the now playing -> channel  tab and see what your noise floor and signal level is, reducing your gain may be enough to help, but I suspect you'll want a better dipole antenna to really separate them, and like I said, get it up as high as you can.  Just remember the center conductor of the coax connects to the top half of the antenna and the shield connects to the bottom half, and it should be mounted vertical.  Space the two halves about 1-2CM apart and make sure you aren't mounting it to anything conductive like wood, I used a strip of Lexan/Plexi but there's other ways to skin that cat too.  The two elements must be totally isolated.

When setting the gain, bring up the waterfall and on the top part, increase the gain just until you see the floor rise significantly (there will usually be one step where it makes a very obvious jump), then back it off one.  That should be a good starting point for gain.  But I'd try and get a better antenna setup first.  You can build one for next to free (may have to buy a piece of SMA to SMA cable and cut an end of to solder on to the copper wires).  Add a magnet ferrite as a choke too, somewhere about 4-6" from where the wire attaches to the antenna.  Wrap the wire through the choke a few times if you can.

I had to do all this with the previous software I was doing so by the time I was toying with SDRTrunk I had it all in place and my gains pretty well figured out.

Zachary3352, 2024-12-18:
Thanks for the suggestions. I lowered the gain and the sample rate, and I'll listen for a day or two to see if it gets any better. If it doesn't, I can upload a couple recordings of calls with the beginning cut off as examples.

I appreciate the antenna tips as well. As I mentioned, this physical setup has worked very well in the past, so it seems unlikely that my antenna is inadequate. The antenna is mounted as high in the room as possible and on an exterior wall. But I'll certainly keep it in mind as a possibility!

Zachary3352, 2024-12-20:
After a couple days, it's somewhat better. Not perfect, but better. I still often miss short calls, but I do feel like I'm receiving more than before.

seth4380, 2024-12-20:
Are any of the p25 conventional channels you guys listing to happen to have encryption on some transmissions? That’s when I notice sdr-trunk missing them. Seems to be right after. For instance to officer is talking on encrypted and the dispatch isn’t. That’s when it seems to miss things. Almost like it gets confused. Then when they are not talking in encrypted it seems to work fine. Sent from my iPhone

Zachary3352, 2024-12-20:
Interesting observation. My town is always in the clear, so I haven't noticed that specifically.

someengineer1, 2024-12-20:
No encryption here.  Occasionally I do see "encrypted" pop up but it is just a bad decode or some extraneous data that got sent as my city is not using encryption at all.

With LSM, gain of 25.x (which puts my noise floor at around -50 and active signals around -10) I get excellent results.

dleiken, 2024-12-28:
I also monitor a few P25 conventional systems with this problem. It seems to be more of a problem for systems with not a lot of activity. I have noticed in the channel spectrum for systems not very active that it does not system locked on the center frequency. Once the channel is active it then finds the center again, but causes a delay and missing part of the transmission. I also have seen a lot of missing transmissions for short calls. Almost seems like the decoder does not have enough time to start decoding the short calls. Maybe a feature to stop the search/drifting can fix this like a NBFM channel that just says locked on the center with not drifting/ searching?

someengineer1, 2024-12-28:
Sounds like you need to get your PPM offset dialed in.  You can't rely on "auto", that's only there to make minor changes due to temperature changes etc.  If it has to do a major adjustment every time a transmission happens that's going to cause it to take longer to lock on.

Note that in between transmissions, there won't be any "lock" or center frequency, you have to wait until there is a live transmission to set your offset to get it centered.

DSheirer, 2026-11-08:
The C4FM decoder was recently updated and should resolve previous issues with decoding.

closed 2026-11-08 by DSheirer