DSheirer/sdrtrunk #2060: LIBUSB_ERROR_NO_DEVICE during get device list

israelgoldstein, 2024-10-30:
**sdrtrunk Version**
SDRTrunk Version  : 0.6.1-beta-2  [8MB/512MB 1%]

**Describe the bug**
Application does not find any SDR's

**To Reproduce**
when i start SDR Trunk (ive tried several builds)
i get this error in the console
ERROR i.g.d.s.t.m.TunerManager - LibUsb - error [-4/LIBUSB_ERROR_NO_DEVICE] during get device list  [25MB/40MB 64%]

**Expected behavior**
im expecting it to find my SDR (or several SDR's that i have attached via USB)

**Screenshots**
2024-10-29 22:59:32.431 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [23MB/40MB 59%]
2024-10-29 22:59:32.509 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [25MB/40MB 62%]
2024-10-29 22:59:32.509 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [25MB/40MB 62%]
2024-10-29 22:59:32.532 ERROR i.g.d.s.t.m.TunerManager - LibUsb - error [-4/LIBUSB_ERROR_NO_DEVICE] during get device list  [25MB/40MB 62%]
2024-10-29 22:59:32.532 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [25MB/40MB 62%]

**Application Log**
2024-10-29 22:59:32.431 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [23MB/40MB 59%]
2024-10-29 22:59:32.509 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [25MB/40MB 62%]
2024-10-29 22:59:32.509 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [25MB/40MB 62%]
2024-10-29 22:59:32.532 ERROR i.g.d.s.t.m.TunerManager - LibUsb - error [-4/LIBUSB_ERROR_NO_DEVICE] during get device list  [25MB/40MB 62%]
2024-10-29 22:59:32.532 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [25MB/40MB 62%]

**Desktop (optional - complete the following information):**
 - OS: Host OS Name:          Windows 11
 - CPU Cores: Host CPU Cores:        4  
 - RAM:Host Max Java Memory:  7 GB  

when i attach same SDRs to my other PC it works, 
this PC also worked fine a few weeks back

DSheirer, 2024-10-30:
On this problematic PC, are you able to use the sdr devices with any other SDR software applications?

You might try reinstalling the drivers with Zadig.  Sometimes Windows updates can break the drivers.

DSheirer, 2024-11-02:
@israelgoldstein Any updates?

closed 2024-11-06 by DSheirer