DSheirer/sdrtrunk #2079: Excessive Event Log Count Can Impact App Performance

DSheirer, 2024-11-03:
**sdrtrunk Version**
Users can turn on event logging and produce a large number of event logs and that can impact application performance.

Consider options:
1. Warn user once the total count or size of event logs exceeds some threshold
2. Investigate options for offloading event log writing and file creation to a separate threaded process.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
3. Click on '....'
4. Scroll down to '....'
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: [e.g. iOS]
 - CPU Cores: [e.g. 4 cores]
 - RAM:[e.g. 16GB]

**Additional context**
Add any other context about the problem here.

kernworks, 2025-01-06:
Does this include application logging?

I was considering writing a PR for #1816 but there wouldn't be much of a point to doing that currently if it could impact application performance.

DSheirer, 2025-01-06:
Application logging is configurable via the logback.xml file.  Logback has an option for a rolling log file that can be configured in this xml file.