DSheirer/sdrtrunk #2081: Add libusb4java-1.3.0-win32-aarch64.jar to allow loading Windows ARM64 version

JVital2013, 2024-11-03:
**sdrtrunk Version**
v0.6.1 beta 3 - Windows ARM64 version

**Describe the bug**
SDRTrunk fails to load because libusb4java-1.3.0-win32-aarch64 is missing.

**To Reproduce**
Attempt to load the ARM64 build on any ARM64-based Windows 11 PC.

**Expected behavior**
libusb4java-1.3.0-win32-aarch64.jar needs to be included with sdrtrunk. Since this variant of libusb4java is not provided by the developer, I've attached a copy I built myself to this post: [libusb4java-1.3.0-win32-aarch64.zip](https://github.com/user-attachments/files/17611952/libusb4java-1.3.0-win32-aarch64.zip)

**Screenshots**
Before adding libusb4java-1.3.0-win32-aarch64.jar:
[image]

After adding libusb4java-1.3.0-win32-aarch64.jar:
[image]

JVital2013, 2024-11-03:
Ah, missed one part - the libusb dll itself also needs to be added into SDRTrunk's `bin` folder. See attached:

[libusb-1.0-win32-aarch64.zip](https://github.com/user-attachments/files/17612012/libusb-1.0-win32-aarch64.zip)

DSheirer, 2024-11-03:
@JVital2013 Thank you very much!  I'll get this added in.

PR #2083 (2024-11-05): #2081 Windows aarch64 libusb support.
Closes #2081 
Adds libusb native libraries to support Windows aarch64 platforms.

Note: the new library bundles both the Windows and OSX native libraries for aarch64 where previously the OSX library was bundled in a separate jar.
closed 2024-11-05 by DSheirer