DSheirer/sdrtrunk #2088: Version 0.6.1 Beta 3

jkraemer21, 2024-11-10:
### Discussed in https://github.com/DSheirer/sdrtrunk/discussions/2073

<div type='discussions-op-text'>

<sup>Originally posted by **DSheirer** November  1, 2024</sup>
## **Application download links for each operating system are available below in the Assets section.**

## Features:
* Tuner minimum and maximum frequency now user adjustable so you can lock a tuner to a specific frequency range
* SDRPlay RSP1B & RSPdxR2 tuners and API 3.15 and 3.16 support
* P25 Phase 1 & 2 expanded message decoding, enhanced traffic channel management, talker alias/gps and ISSI/Roam fully qualified Identifiers with alias support
* DMR significant improvements to processing and talker alias/gps support
* Enhanced map display
* Radio Reference API version 18
* Fix for some critical issues causing out-of-memory, application crashes, and channels stuck in TEARDOWN
* Automatic threading dead-lock detection, notification and report creation
* Create diagnostic reports (File >> Reports) for reporting out-of-memory and app crash errors
* Adds Windows aarch64 support (experimental)
* Latest Java OpenJDK 23

## Requirements:
* Windows x86/aarch64 (64-bit)
* Linux x86/aarch64 (64-bit)
* OSX x86/aarch64 (64-bit)

**Note**: Linux for arm-32 is NOT supported

## Optional Requirements
**JMBE version 1.0.9+** is required for decoding APCO25 Phase 1/2 and DMR digital audio on Linux and OSX aarch64 and this library can be compiled within the sdrtrunk application.

Legacy instructions for compiling the JMBE library are located at: **https://github.com/DSheirer/sdrtrunk/wiki/JMBE**

## User Manual
* [Wiki](https://github.com/DSheirer/sdrtrunk/wiki)
* [User Manual](https://github.com/DSheirer/sdrtrunk/wiki/User-Manual)

## Support
* [Support](https://github.com/DSheirer/sdrtrunk/wiki/Support)

## What's Changed
* #2025 P25P1 Audio Module Null Codec Error. by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2027
* #2030 Playlist Editor Double-Click Channel Start by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2031
* #2032 Audio Output Logs Failed Attempt To Update Gain/Mute Controls by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2033
* #2035 P25P1 Harris Patched Talkgroup Values Incorrect by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2036
* #2028 Windows ARM/aarch64 Operating System Support by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2029
* #1060 Alias List Delete and Rename Operations To Playlist Editor by @whelgeson in https://github.com/DSheirer/sdrtrunk/pull/2038
* #2039 P25 Phase 2 Incorrect Talkgroup & Radio Values In Call Metadata by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2042
* #2043 DMR Tier 3 Capacity Max Incorrect Talkgroup Values by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2044
* #2045 P25 Phase 1 Data Channels Always Ignored by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2046
* #2037 JavaFX Windows Can't Be Shrunken by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2047
* #2040 Reset AMBE/IMBE Audio Codec After Each Call by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2048
* #2034 Map Plotted Elements Timestamp Includes Date by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2049
* #1867 Event Logger - Log Radio ID and Not Network Radio Alias by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2050
* #2055 Multi-Frequency channel source with channel rotation monitor st… by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2056
* #1992 Traffic channels stuck in teardown. by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2061
* #2062 Verbose logging option for Broadcastify Feed to troubleshoot st… by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2063
* #2064 Moto talker GPS event TO/FROM identifiers. by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2065
* #2066 P25 Phase 1 Frequency Band Link Control Messages Checked For Valid CRC by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2067
* #2068 P25 traffic channel manager ignores traffic channel requests if… by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2069
* #2055 MultiFrequencyTunerChannelSource Null Pointer Error On Channel … by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2070
* #2071 Update To Version 0.6.1 Beta 3 by @DSheirer in https://github.com/DSheirer/sdrtrunk/pull/2072

**Full Changelog**: https://github.com/DSheirer/sdrtrunk/compare/v0.6.1-beta-2...v0.6.1-beta-3

<hr /><em>This discussion was created from the release <a href='https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.1-beta-3'>Version 0.6.1 Beta 3</a>.</em></div>

jkraemer21, 2024-11-10:
https://www.radioreference.com/db/site/42816
Phase 1 CC with Phase 2 Voice Channels. Unable to setup channel.

DSheirer, 2024-11-10:
> https://www.radioreference.com/db/site/42816 Phase 1 CC with Phase 2 Voice Channels. Unable to setup channel.

I just tested creating a channel configuration for that site using the Radio Reference tab in the Playlist Editor and it's working correctly. 

You'll need to provide more details ... how are you trying to setup the channel?  What errors are you seeing?  Are you able to start the channel?

jkraemer21, 2024-11-10:
The channel starts but no voice following. I have tried every combination for channel setup from RR but still no success.

DSheirer, 2024-11-10:
Can you post a screenshot of the application that shows the now playing window with the control channel selected, and the Messages tab below selected and showing the decoded control channel messaging?

There's a screen shot menu item that you can use.

jkraemer21, 2024-11-10:
Here is the Messages tab and the Status page. Please don't hesitate to ask if you require any more information.
<https://1drv.ms/i/c/56b5eb1931060221/ESes-LxwGW5JnY3llS72wl4BRbrjAO9TKRpO5V0JlV0hjQ>
[https://southcentralus1-mediap.svc.ms/transform/thumbnail?provider=spo&farmid=193110&inputFormat=png&cs=NWQ2NjE5NTAtMzQ3NS00MWNkLWEyYzMtZDY3MWEzMTYyYmMxfFNQTw&docid=https%3A%2F%2Fmy.microsoftpersonalcontent.com%2F_api%2Fv2.0%2Fdrives%2Fb!38bE4L1afkCdI-iMHs9qfQ0TFhkb4VdGmKrMZJM5tsH7zpMSo5YoSJepPnFTAmbV%2Fitems%2F01QTXJSMBHVT4LY4AZNZEZ3DPFSUXPNQS6%3Ftempauth%3Dv1e.eyJzaXRlaWQiOiJlMGM0YzZkZi01YWJkLTQwN2UtOWQyMy1lODhjMWVjZjZhN2QiLCJhcHBpZCI6IjVkNjYxOTUwLTM0NzUtNDFjZC1hMmMzLWQ2NzFhMzE2MmJjMSIsImF1ZCI6IjAwMDAwMDAzLTAwMDAtMGZmMS1jZTAwLTAwMDAwMDAwMDAwMC9teS5taWNyb3NvZnRwZXJzb25hbGNvbnRlbnQuY29tQDkxODgwNDBkLTZjNjctNGM1Yi1iMTEyLTM2YTMwNGI2NmRhZCIsImV4cCI6IjE3MzEyODMyMDAifQ.sxGY6lej1T-ye4VouFkbxCEw7MmWbHiVgweNa5vprS9Vsy6UJ_MGVI0NuWmCn6Ge_GkIl9_52MbqTGqcrG8T9-b1j0xlX1WJTSw4aP30Fcvd8bijOi7Tb4rMEbnvJs1vnfedPGa0UVIpnx60HVuaE_GIoQ4i_oAnYHaiPuuyWNKTs1lMR9XPvUSzhlJQfWewKn-0aPkLhMs9MT-MTlg7IUnhzNwH_HyzRiGO3gUjt8kKlxZ0wSVX6d1orRv-1-PrstnIYeeeXW3Cvm2scMjtlRG5sU9UDklUDinOwvARWU4iFTK-aM-P0XFMGmGPP85OnPpQmhXvuDN-oBN6JmyyZYNQCRsL6Vu1fgevPgzdqRXwulNIS3tzU9Z1jOrJ0Iv_nDqaZ_vy7XLBsGQG01xgolLdbS0S_Mfx9-7p1bA_0jw.erAUsMBif824PWisYPn4AAiRRcyxbJzwMR0LbmaMtmQ%26version%3DPublished&width=176&height=176&cb=63866859528]
[https://res.cdn.office.net/assets/mail/file-icon/png/cloud_blue_16x16.png]Screenshot 2024-11-10 131027.png
[https://res.public.onecdn.static.microsoft/assets/mail/file-icon/png/txt_16x16.png]Activity Summary - DecoderP25 Phase.txt<https://1drv.ms/t/c/56b5eb1931060221/ESIUrWSVJHhNgO8k3g3qLcUBPgdxhT4oiHdakvJNlxL39A>

​

DSheirer, 2024-11-10:
Looks like you're getting good control channel decode.

What's the events tab look like?  Are you seeing traffic channel grants?  Can you post a screenshot with the Events tab showing and the control channel selected in the now playing window?

jkraemer21, 2024-11-10:
Yes, channel grants are there.

<https://1drv.ms/i/c/56b5eb1931060221/EUjCvmHCF1lIhkoaIW4WdrYBt9O1zI2ndJrKRPXDJSKKBg>
[https://southcentralus1-mediap.svc.ms/transform/thumbnail?provider=spo&farmid=193110&inputFormat=png&cs=NWQ2NjE5NTAtMzQ3NS00MWNkLWEyYzMtZDY3MWEzMTYyYmMxfFNQTw&docid=https%3A%2F%2Fmy.microsoftpersonalcontent.com%2F_api%2Fv2.0%2Fdrives%2Fb!38bE4L1afkCdI-iMHs9qfQ0TFhkb4VdGmKrMZJM5tsH7zpMSo5YoSJepPnFTAmbV%2Fitems%2F01QTXJSMCIYK7GDQQXLFEIMSQ2EFXBM5VW%3Ftempauth%3Dv1e.eyJzaXRlaWQiOiJlMGM0YzZkZi01YWJkLTQwN2UtOWQyMy1lODhjMWVjZjZhN2QiLCJhcHBpZCI6IjVkNjYxOTUwLTM0NzUtNDFjZC1hMmMzLWQ2NzFhMzE2MmJjMSIsImF1ZCI6IjAwMDAwMDAzLTAwMDAtMGZmMS1jZTAwLTAwMDAwMDAwMDAwMC9teS5taWNyb3NvZnRwZXJzb25hbGNvbnRlbnQuY29tQDkxODgwNDBkLTZjNjctNGM1Yi1iMTEyLTM2YTMwNGI2NmRhZCIsImV4cCI6IjE3MzEyODMyMDAifQ.K3qnHhFcEIthVVndTfouIHTdQDmrXcl9DcrIH2XwUkehHYI6189ULmEXaZf6DXUGG6qePW7OjvyF0PDswB670KLWJrVWTwYeVG9Eo3_C_Bb6UWz_aDeouxIXzO6AD0uz0PvztPTX06LN3OZU_eyhLiDec2ohR77zG6CrsbyLA4VGdU8Py2hZdgdmdRTjnF4zV6wRoEQUTulCJzoLAA1CJN8TFrzb5jFmS8YKEFYVb2Cg4bbz0ilcQNViBwrNLeBz4DD6INla7T-_Lpp9JVlli1pW2RV-j85wwkzyw3-LGW0h6MKX0InqgbARNw8pRMSzC5yqvdWd8y3_avxCKT9Vmlwh6iyU-0UF3elSrcJ7160umBqOnaN2mFORIMDLRfc1rMy_g0c10-NugygstZhkm1UKIvg_2TRbWxaFt-g6ktzRygzRx1d_5KcqPQAm5kyc.5ecD0wviw49H_YqMAvOLCavXmJLqEVu4xOFhqUtIzG4%26version%3DPublished&width=176&height=176&cb=63866868779]
[https://res.cdn.office.net/assets/mail/file-icon/png/cloud_blue_16x16.png]Screenshot 2024-11-10 154915.png

closed 2026-11-16 by DSheirer