DSheirer/sdrtrunk #2091: Unable to import talkgroups from Radio Reference to an Alias list in sdrtrunk

bhall7, 2024-11-12:
Version: sdrtrunk nightly (11-12-2024)

I'm unable to import talkgroups from Radio Reference to an alias list. When I select my alias list and click "Import All Talkgroups," nothing happens. I've attached a screen capture that demonstrates the issue.

I also tried deleting the alias list and creating a new one, but I am still unable to import any talkgroups from Radio Reference into this group. I can manually create new talkgroups, just can't import any from RR.

**To Reproduce**
Steps to reproduce the behavior:
1. Radio Reference tab in Playlist Editor and pick a system
2. Click on Talkgroup View, and select an alias list to import to next to the drop-down "Import To Alias List"
3. Click "Import All Talkgroups"
4. Note that none of the talkgroups get imported to the specified alias list

**Expected behavior**
I would expect all the talkgroups to be imported to the specified alias list.

Application log and screen capture of the behavior uploaded.

**Desktop (optional - complete the following information):**
 - OS: Windows 11 Pro
 - CPU Cores: 4 cores
 - RAM: 16GB

[sdrtrunk_app.log.txt](https://github.com/user-attachments/files/17722903/sdrtrunk_app.log.txt)

https://github.com/user-attachments/assets/cbf56053-0303-4e21-a5db-8a4b07018a6a

Looks like others have reported this issue on the SDRTrunk group.

DSheirer, 2024-11-13:
You've already aliased those talkgroups to the 'All' alias.

The default behavior of 'Import All Talkgroups' button is to not replace anything that you're already aliased so that it doesn't overwrite anything that you may have customized.

You can temporarily remove the Talkgroup Range from the All Alias, import your talkgroups, and then restore the All alias.

closed 2024-11-13 by DSheirer

bhall7, 2024-11-13:
Unfortunately, even after deleting the "All" alias, and the alias list is completely empty, nothing is getting imported. See the screen capture here.

https://github.com/user-attachments/assets/7ac75be5-4b40-48cf-b975-3dc57b8db2f8

bhall7, 2024-11-22:
Looks like this was fixed in the nightly build from 11-12-2024. 👍