DSheirer/sdrtrunk #2103: Error running v0.6.1 on Arch Linux (UnsupportedClassVersionError)

k4yt3x, 2024-12-03:
**sdrtrunk Version**
v0.6.1

**Describe the bug**
I am the maintainer of the `sdrtrunk-bin` package on Arch Linux AUR. I noticed that a new version was release so I tried to package the new version. This version does not seem to run out of the box with the bundled Java runtime. it produces this error:

> Exception in thread "main" java.lang.UnsupportedClassVersionError: com/github/dsheirer/sdrplay/api/v3_07/sdrplay_api_h (class file version 63.65535) was compiled with preview features that are unsupported. This version of the Java Runtime only recognizes preview features for class file version 67.65535

I've also tried the system OpenJDK 23, and it seems to be missing dependencies to run sdrtrunk. The runtime version seems to support class file version higher than the program's version, which doesn't make sense to me. Could you please take a look to see if the attached Java binary is of the wrong version or if I have done something wrong here? Thanks.

**To Reproduce**

Build and run sdrtrunk on Arch Linux with this PKGBUILD:

```pkgbuild
pkgname=sdrtrunk-bin
pkgver=0.6.1
pkgrel=1
pkgdesc='A cross-platform java application for decoding, monitoring, recording and streaming trunked mobile and related radio protocols using SDR'
arch=('x86_64' 'aarch64')
url='https://github.com/DSheirer/sdrtrunk'
license=('GPL3')
depends=('at-spi2-core' 'libnet' 'alsa-lib' 'gtk3')
provides=('sdrtrunk')
conflicts=('sdrtrunk' 'sdrtrunk-git' 'sdrtrunk-bin')

source_x86_64=("${url}/releases/download/v${pkgver}/sdr-trunk-linux-x86_64-v${pkgver}.zip"
               "sdrtrunk.desktop")

source_aarch64=("${url}/releases/download/v${pkgver}/sdr-trunk-linux-aarch64-v${pkgver}.zip"
                "sdrtrunk.desktop")

b2sums_x86_64=('36a1b31b18ee14cc4df0472ff04fc199864e0d69f34f86908e8e0a35d57793f2c2388e214e9b6fa29237c9b8f18f04b96d908975da8e268a3c9503ff108a1455'
               '8e5ae51bf3acf88582913d8c1c120ce2ecc5082d0f74abe6f4cfeefc9222b1507ce18de089e5d7fd64f2cf80ce98be95cdc64629f5ebc2aaa1bbb34e9d9bf306')

b2sums_aarch64=('7d67d4b5e77b649714335acdf74d76727a87ee899585c6b7038b712858ecbb779abce39ae0a25bf3d58425d2774d96572fc66f6f3009652abbb480fa31bc2cc8'
                '8e5ae51bf3acf88582913d8c1c120ce2ecc5082d0f74abe6f4cfeefc9222b1507ce18de089e5d7fd64f2cf80ce98be95cdc64629f5ebc2aaa1bbb34e9d9bf306')

package() {
    install -Dm644 sdrtrunk.desktop "${pkgdir}/usr/share/applications/sdrtrunk.desktop"

    cd "sdr-trunk-linux-${CARCH}-v${pkgver}"
    mkdir -p "${pkgdir}/usr/share/sdrtrunk" "${pkgdir}/usr/bin"
    cp -ra * "${pkgdir}/usr/share/sdrtrunk"

    ln -s "/usr/share/sdrtrunk/bin/sdr-trunk" "${pkgdir}/usr/bin/sdr-trunk"
}
```

**Expected behavior**
The application starts and runs normally.

**Actual behavior**
The application fails with this error:

```console
Exception in thread "main" java.lang.UnsupportedClassVersionError: com/github/dsheirer/sdrplay/api/v3_07/sdrplay_api_h (class file version 63.65535) was compiled with preview features that are unsupported. This version of the Java Runtime only recognizes preview features for class file version 67.65535
        at java.base/java.lang.ClassLoader.defineClass1(Native Method)
        at java.base/java.lang.ClassLoader.defineClass(Unknown Source)
        at java.base/java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(Unknown Source)
        at java.base/jdk.internal.loader.BuiltinClassLoader.findClassOnClassPathOrNull(Unknown Source)
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClassOrNull(Unknown Source)
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
        at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
        at com.github.dsheirer.sdrplay.device.RspDuoMode.<clinit>(RspDuoMode.java:29)
        at com.github.dsheirer.sdrplay.DeviceSelectionMode.<clinit>(DeviceSelectionMode.java:32)
        at io.github.dsheirer.source.tuner.TunerFactory$1.<clinit>(TunerFactory.java:214)
        at io.github.dsheirer.source.tuner.TunerFactory.getUsbTuner(TunerFactory.java:292)
        at io.github.dsheirer.source.tuner.manager.DiscoveredUSBTuner.start(DiscoveredUSBTuner.java:107)
        at io.github.dsheirer.source.tuner.manager.TunerManager.tunerStatusUpdated(TunerManager.java:382)
        at io.github.dsheirer.source.tuner.manager.TunerManager.startAndConfigureTuner(TunerManager.java:286)
        at io.github.dsheirer.source.tuner.manager.TunerManager.addUsbTuner(TunerManager.java:254)
        at io.github.dsheirer.source.tuner.manager.TunerManager.discoverUSBTuners(TunerManager.java:241)
        at io.github.dsheirer.source.tuner.manager.TunerManager.start(TunerManager.java:126)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:177)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:731)
```

**Desktop (optional - complete the following information):**
 - OS: Arch Linux 6.12.1-arch1-1
 - CPU Cores: 16C/32T
 - RAM: 64 GiB

DSheirer, 2024-12-06:
@k4yt3x Is there any chance that there's remnants of a previous sdrtrunk install and you're unzipping on top of it?  Usually when you have these Unsupported Class Version errors, it means that some of the code was compiled with a different compiler process.

k4yt3x, 2024-12-06:
You are absolutely right! Theoretically it should remove everything before the new install, but somehow some sdrplay jar files and stuff were left in the directory. After removing them it works fine.

Thanks for the help. I've pushed the new version to the AUR.

closed 2024-12-06 by k4yt3x