DSheirer/sdrtrunk #2107: SDR Trunk Version 0.6.1 (final) Issues.

Technology-Experienced-Expert-Wizard, 2024-12-07:
**sdrtrunk Version**
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.
Version 0.6.1 (final)
**Describe the bug**
A clear and concise description of what the bug is. 
I have added alias lists with all desired talkgroup numbers  I want to listen to and  I'm receiving talkgroup i dont what to listen to that arent in my alias list..

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected to happen  to only have programed talkgroups in alias list to only get received/ listen to  only.   

**Screenshots**
If applicable, add screenshots to help explain your problem.
N/A
**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.
N/A

**Desktop (optional - complete the following information):**
 - OS: [e.g. iOS]
 - CPU Cores: [e.g. 4 cores]
 - RAM:[e.g. 16GB]

**Additional context**
Add any other context about the problem here.

Crash0v3r1de, 2024-12-27:
Add the talk group you don't want to hear and turn off listen
[image]

Technology-Experienced-Expert-Wizard, 2024-12-28:
The issue with that turning off listen option is the talkgroups that have
listen off, these talkgroups still show up in sdr trunk call screen but no
audio because listen is off for specific talkgroups and the talkgroups that
have listened off should not  show up on sdr calls screen but they do show
up but they shouldn't  show up.  kindly make this  sdr trunk change  to
 kindly make talkgroups with listen off that dont show up in the sdr trunk
call screen.
Thank you.

ᐧ

Crash0v3r1de, 2024-12-28:
> The issue with that turning off listen option is the talkgroups that have
> 
> listen off, these talkgroups still show up in sdr trunk call screen but no
> 
> audio because listen is off for specific talkgroups and the talkgroups that
> 
> have listened off should not  show up on sdr calls screen but they do show
> 
> up but they shouldn't  show up.  kindly make this  sdr trunk change  to
> 
>  kindly make talkgroups with listen off that dont show up in the sdr trunk
> 
> call screen.
> 
> Thank you.
> 
> 
> 
> 
> 
> 
> 
> > Add the talk group you don't want to hear and turn off listen
> 
> > image.png (view on web)
> 
> > <https://igafnl.com/click?redirect=https%3A%2F%2Fgithub.com%2Fuser-attachments%2Fassets%2F932bcb4b-073b-4e98-a4ec-001532307c8c&dID=1735413857450&hashId=8f35d0e109b66f8f52f7a52f664ee5dbe89e557250878f8bf3f82fe6cc92e938cf6c&linkName=image.png
> 
> > (view on web)>
> 
> >
> 
> 
> 
> > <https://igafnl.com/click?redirect=https%3A%2F%2Fgithub.com%2FDSheirer%2Fsdrtrunk%2Fissues%2F2107%23issuecomment-2563831242&dID=1735413857450&hashId=8f35d0e109b66f8f52f7a52f664ee5dbe89e557250878f8bf3f82fe6cc92e938cf6c&linkName=view
> 
> 
> 
> 
> 
> 
> >
> 
> ᐧ

Without you making a code addition to add this into the user preferences and then adding the logic to not show the tall groups currently that's not possible.

This is a public community repo so you're always able to make your changes and submit a PR. The tower is going to output its talk groups and function like normal so it being in the call log from my point of view makes perfect sense.

Hope you like to code in Java!

DSheirer, 2026-11-08:
Create a talkgroup range as a catch-all for all non-aliased talkgroups and set it to do not listen.  The calls that you're hearing are from non-aliased talkgroups and you need a rule for how to handle those.  The default rule for non-aliased talkgroups is: listen:yes.

closed 2026-11-08 by DSheirer