DSheirer/sdrtrunk #2109: Audio stops working after a few hours. (0.6.1 Final)

LordDonkey, 2024-12-14:
After a few hours of operation, SDRTrunk stops playing audio. I see that calls are happening and are still being recorded, but nothing is playing back. If I go into SDRTrunk preferences and change the audio devices around, sound starts working again for awhile. I've included a thread dump and processing diagnostic report if that helps.

I've received this error in the command window: `2024-12-13 11:12:20.638 ERROR i.g.d.m.d.p.a.P25P2AudioModule - Error synthesizing AMBE audio - continuing [Cannot invoke "io.github.dsheirer.sample.Listener.receive(Object)" because "this.mMessageListener" is null [291MB/416MB 69%]`

[20241214_110513_sdrtrunk_thread_dump.log](https://github.com/user-attachments/files/18136619/20241214_110513_sdrtrunk_thread_dump.log)
[20241214_110509_sdrtrunk_processing_diagnostic_report.log](https://github.com/user-attachments/files/18136620/20241214_110509_sdrtrunk_processing_diagnostic_report.log)

IMG20191023, 2024-12-19:
Same issue here, I switch the audio output to a different device then back and it beings to work again.

FeedMePizza, 2024-12-22:
Same issue here with both 0.6.0 and 0.6.1.  It only happens with USB speakers.  The audio doesn't stop if I plug analog speakers into the line-out/speaker port on my PC.

I don't see any errors but perhaps I don't have logging setup properly.

Thanks for the "switch the audio output to a different device then back and it begins to work again" trick.  I had been restarting SDRTrunk to regain audio.

I'm running Debian GNU/Linux 12 (bookworm).

svdasein, 2025-10-07:
Same issue here.  

```
System:
  Kernel: 6.8.0-83-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0
  Desktop: Cinnamon v: 6.4.8 Distro: Linux Mint 22.1 Xia
    base: Ubuntu 24.04 noble
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio driver: snd_hda_intel
    v: kernel bus-ID: 04:00.1
  Device-2: AMD ACP/ACP3X/ACP6x Audio Coprocessor driver: snd_rn_pci_acp3x
    v: kernel bus-ID: 04:00.5
  Device-3: AMD Family 17h/19h HD Audio driver: snd_hda_intel v: kernel
    bus-ID: 04:00.6
  Device-4: Jieli USB Composite Device driver: snd-usb-audio type: USB
    bus-ID: 1-3.2:7
  Device-5: Linux Foundation NewEye 61 Camera driver: snd-usb-audio,uvcvideo
    type: USB bus-ID: 3-2.1.1:7
  API: ALSA v: k6.8.0-83-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: n/a (root, process)
Info:
  Memory: total: 28 GiB available: 27.33 GiB used: 8.49 GiB (31.1%)
  Processes: 457 Uptime: 17d 21h 55m Init: systemd target: graphical (5)
  Packages: 2856 Compilers: gcc: 13.3.0 Shell: Bash v: 5.2.21 inxi: 3.3.34
```

The switch audio output trick is only hit or miss for me.  I've tried the nightly - same issue (maybe even slightly worse).

DSheirer, 2026-02-08:
This should be resolved.  Additional enhancements under #2356

closed 2026-02-08 by DSheirer