DSheirer/sdrtrunk #2132: version 6.1 appears not to honor "Max traffic channels", causes max channels exceeded and no output.

wsbush, 2025-01-17:
**sdrtrunk Version**
Master Branch

**Describe the bug**

I have both 6.0 release and 6.1 release loaded.  
Channel used has "6" as max traffic channels.
Monitoring "Tuners/Channels", running 6.0, i see locked channels increasing/decreasing as needed, with a maximum of 4 in my case.  "Now Playing/Events show no issues. Systems running fine.   With 6.1, and monitoring "Tuners/channels" , locked channels do not exceed 2, "Now Playing/Events" show max channels exceeded messages.  within one to two minutes, output ceases, locked channels = 1, and "control" channel has a channel of 0-1429 in it ( dont know if that matters)

There are no errors logged.

**To Reproduce**
ive defined the criteria in the description above.  

**Expected behavior**
honor max traffic channels

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
BELOW ARE LOGS FOR BOTH 6.0 and 6.1

20250117 114407.868 [main] INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0  [14MB/60MB 24%]
20250117 114407.869 [main] INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.2  [14MB/60MB 24%]
20250117 114407.870 [main] INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-12-16T06:54:18.883-0500  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Build-JDK         : 20.0.1 (BellSoft 20.0.1+10  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.2.0-37-generic  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog -   [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 7  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Host OS Version:       6.1  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  958 MB  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog - Storage Directories:  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\NewAdmin\SDRTrunk  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\NewAdmin\SDRTrunk\logs  [14MB/60MB 24%]
20250117 114407.871 [main] INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\NewAdmin\SDRTrunk\event_logs  [14MB/60MB 24%]
20250117 114407.888 [main] INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\NewAdmin\SDRTrunk\playlist  [14MB/60MB 24%]
20250117 114407.890 [main] INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\NewAdmin\SDRTrunk\recordings  [14MB/60MB 24%]
20250117 114407.915 [main] INFO  i.g.dsheirer.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [15MB/60MB 25%]
20250117 114407.918 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\NewAdmin\SDRTrunk\SDRTrunk.properties]  [15MB/60MB 25%]
20250117 114408.336 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [16MB/27MB 60%]
20250117 114408.498 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.262  [17MB/27MB 64%]
20250117 114408.498 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.22.11312  [17MB/27MB 64%]
20250117 114408.812 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [7] potential usb devices  [17MB/27MB 64%]
20250117 114408.831 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [2] Port [1] Tuner Class [RTL-2832]  [17MB/27MB 64%]
20250117 114408.837 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:1 - Added / Starting ...  [17MB/27MB 66%]
20250117 114410.119 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [7MB/27MB 28%]
20250117 114410.299 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [8MB/27MB 30%]
20250117 114410.328 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [8MB/27MB 31%]
20250117 114410.441 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [10MB/27MB 39%]
20250117 114410.455 [main] WARN  i.g.d.s.t.sdrplay.api.Version - Unrecognized SDRplay API version [3.15]  [10MB/27MB 39%]
20250117 114410.456 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library is not available.  [10MB/27MB 39%]
20250117 114410.463 [main] INFO  i.g.d.settings.SettingsManager - SettingsManager - loading settings file [C:\Users\NewAdmin\SDRTrunk\settings\settings.xml]  [11MB/27MB 40%]
20250117 114411.075 [main] WARN  i.g.d.vector.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [15MB/27MB 55%]
20250117 114412.121 [main] INFO  i.g.d.playlist.PlaylistManager - Loading playlist [C:\Users\NewAdmin\SDRTrunk\playlist\Memphis Police.xml]  [15MB/27MB 56%]
20250117 114412.650 [main] INFO  i.github.dsheirer.gui.SDRTrunk - starting main application gui  [18MB/27MB 70%]
20250117 114424.211 [sdrtrunk scheduled thread 3] INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] now - timer invoked  [23MB/44MB 53%]
20250117 114424.336 [sdrtrunk scheduled thread 3] INFO  i.g.d.a.c.mbe.JmbeAudioModule - Loading JMBE library from [C:\Users\NewAdmin\SDRTrunk\jmbe\jmbe-1.0.9.jar]  [27MB/45MB 61%]
20250117 114424.349 [sdrtrunk scheduled thread 3] INFO  i.g.d.a.c.mbe.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [28MB/45MB 62%]
20250117 114424.351 [sdrtrunk scheduled thread 3] INFO  i.g.d.a.c.mbe.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [28MB/45MB 62%]
20250117 114424.389 [sdrtrunk scheduled thread 3] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [29MB/45MB 65%]
20250117 114425.258 [sdrtrunk polyphase channel thread 1] INFO  i.g.d.a.c.mbe.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [92MB/257MB 35%]
20250117 120346.871 [AWT-EventQueue-0] INFO  i.github.dsheirer.gui.SDRTrunk - Application shutdown started ...  [196MB/442MB 44%]
[... 37 lines cut ...]
20250117 120419.244 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [2] Port [1] Tuner Class [RTL-2832]  [13MB/24MB 57%]
20250117 120419.250 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:1 - Added / Starting ...  [13MB/24MB 57%]
20250117 120420.695 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [8MB/24MB 34%]
20250117 120420.882 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [8MB/24MB 35%]
20250117 120421.059 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library v3.15 - loaded  [10MB/24MB 45%]
20250117 120421.238 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered [0] RSP devices from SDRplay API  [11MB/24MB 48%]
20250117 120421.253 [main] INFO  i.g.d.settings.SettingsManager - SettingsManager - loading settings file [C:\Users\NewAdmin\SDRTrunk\settings\settings.xml]  [11MB/24MB 49%]
20250117 120421.508 [main] INFO  i.g.d.m.DiagnosticMonitor - Diagnostic monitoring enabled running every 30 seconds  [16MB/24MB 70%]
[... cut ...]
20250117 120432.450 [AWT-EventQueue-0] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [37MB/44MB 84%]
20250117 120436.389 [sdrtrunk channel [1/P25-1] 859937500 thread 1] INFO  i.g.d.a.c.mbe.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [66MB/117MB 56%]

**Desktop (optional - complete the following information):**
 - OS: Windows 7
 - CPU Cores: 4
 - RAM: 4 GB

**Additional context**
Add any other context about the problem here.

Randesign, 2025-01-18:
I just tried this and can't verify. I unplugged all but one tuner. Set my max traffic channels to 6 (and then tried 4) and tuned to a busy P25 site and I've been running for 30 minutes without issue. The only problem I'm seeing is that I don't have enough tuners to cover the spectrum needed to monitor this site. In both cases of setting max traffic channels to either 6 or 4, it's showing up to the max locked channels (+1 to include the control channel).

You can enable the call event logs by going to the bottom of the Channels tab in the playlist. 

I can't duplicate your OS or memory. So you might also go under the File menu and select the reports. 

I also just noticed you're looking at a phase 2 site. I don't have access to one of those.

wsbush, 2025-01-18:
It’s tough to duplicate I’m sure.  I think work has been done for memory
mgmt so I was thinking it was somewhere in that code area.

As an aside, I upgraded to 6.1 because I bought an SDRPlay rsdxr2.  Using
it with my old laptop it worked for a short time then maxed memory and it
crashed.  Apparently the sdrplay api is a memory hog.  I’m going to add
memory to 16 GB to see how it works.

Really appreciate you taking a look at it.

Thanks, Scott

Jim-88, 2025-01-29:
I'm new to SDRTrunk.  So I don't know if this is related.  I noticed in 0.61 final while monitoring a P25 P2 700Mhz system that SDRTrunk never uses more than 2 tuners (RTL-SDR) even though I have 3.  The system is only 4 Mhz wide.  So 2 tuners could cover the range of frequencies used, but one tuner always seems to take its half out of the middle.  The second tuner seems to only get used occasionally and switches between high and low frequencies while the 3rd tuner is ignored unless I monitor a second system at the same time.  So I know it works.  The control frequency is in the middle of the range.  Seems like there is an opportunity for some improvement in how the tuners are allocated.

liquidmayo, 2025-01-29:
Just to add in, I have a system setup, 3 rtlsdr and 2 RSP1B - Monitoring 7
control channels of p25 systems.  On the FINAL version, one of the
counties, who decided not to use MARCS, but their own p25 system, started
having this MAX TRAFFIC CHANNELS EXCEEDED issue. Increasing the max channel
value in the settings just delays it eventually happening as described. It
only happens to this one system, while not impacting the 6 others.

Randesign, 2025-02-03:
@liquidmayo It would be nice if you could provide more details, like an event log file for the county system having the issue. 

Playlist Editor -> Channels -> Logging -> Traffic Channel Call Events  and perhaps Call Events. You should be able to attach a log file to one of these "issue messages".

Randesign, 2025-02-03:
@liquidmayo Is this also a phase 2 system?

wsbush, 2025-02-03:
Its a p25 phase 1.   Ive attached log plus event logs requested.  Runs
about a minute before it quits.

Randesign, 2025-02-03:
@wsbush no logs came through. I think you can't attach files via email. You have to use the github website interface. In the lower left corner of the comment box it says "Paste, drop or click to add files".

wsbush, 2025-02-03:
@Randesign   just added below.

[20250203_114207.920_859937500_Hz_Control_call_events.log](https://github.com/user-attachments/files/18648967/20250203_114207.920_859937500_Hz_Control_call_events.log)
[20250203_114213.376_857712500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18648968/20250203_114213.376_857712500_Hz_T-Control_call_events.log)

[sdrtrunk_app.log](https://github.com/user-attachments/files/18648985/sdrtrunk_app.log)

wsbush, 2025-02-04:
> Your traffic channels are P25 phase 2.
>
> I may have found the problem and it’s related to phase 2 traffic channels
> only. Since I can’t test and verify phase 2 without driving somewhere, I’ll
> just leave this here for @DSheirer <https://github.com/DSheirer> to look
> at.
>
> In P25TrafficChannelManager the following method definition looks
> incorrect (line 251):
>
> private void createPhase2TrafficChannels(int trafficChannelPoolSize,
> DecodeConfiguration decodeConfiguration)
>
> And should be:
>
> private void createPhase2TrafficChannels(int trafficChannelPoolSize,
> DecodeConfigP25Phase2 decodeConfiguration)

I see. Good find.  Thanks for looking into it.

Thanks, Scott

Randesign, 2025-02-09:
@wsbush That wasn't it. I had an occasion to run an errand near a busy phase 2 system. I was able to verify this bug exists. The system had a fair number of TDMA data traffic as well as voice traffic, similar to the system in your log file. I only had about 15 minutes available so my testing wasn't thorough. I'll continue looking at the code, but I'll have to leave this to someone else who has better access to a phase 2 system. You might try switching "ignore data channels" to "on" to see if that changes things.

wsbush, 2025-02-09:
Thanks for the tip and looking into this.

scramblr, 2025-02-24:
I can confirm I had this issue as well, and that the only version in releases that works is Beta-1 - Can you try this version and let me know if it works for you? [https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.1-beta-1](https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.1-beta-1) - Can you try this version and see if it works for you?

wsbush, 2025-02-24:
> scramblr

Will, do, thanks.

Scott Bush
***@***.***

wsbush, 2025-02-24:
Progress.   Using an RTL-SDR, system didnt hang.   log file is going nuts with amount of traffic missing because of too few sdr's.  FYI my SDRPlay RSPdx isnt working with beta but confirmed still works with 6.1  Beta doesnt recognize API version.  

20250224 123846.065 [main] WARN  i.g.d.s.t.sdrplay.api.Version - Unrecognized SDRplay API version [3.15]  [10MB/28MB 38%]
20250224 123846.065 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library is not available.  [10MB/28MB 38%]

ive attached log files.

Thanks

[sdrtrunk_app.log](https://github.com/user-attachments/files/18948630/sdrtrunk_app.log)

[20250224_125229.512_859937500_Hz_Control_call_events.log](https://github.com/user-attachments/files/18948635/20250224_125229.512_859937500_Hz_Control_call_events.log)
[20250224_125418.613_857962500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948638/20250224_125418.613_857962500_Hz_T-Control_call_events.log)
[20250224_125327.255_857712500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948636/20250224_125327.255_857712500_Hz_T-Control_call_events.log)
[20250224_125239.861_858962500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948637/20250224_125239.861_858962500_Hz_T-Control_call_events.log)
[20250224_125243.162_857962500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948640/20250224_125243.162_857962500_Hz_T-Control_call_events.log)
[20250224_125230.406_857937500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948641/20250224_125230.406_857937500_Hz_T-Control_call_events.log)
[20250224_125234.867_858962500_Hz_T-Control_call_events.log](https://github.com/user-attachments/files/18948639/20250224_125234.867_858962500_Hz_T-Control_call_events.log)

scramblr, 2025-03-01:
So for clarification, downgrading to Beta-1 worked? I'm not seeing any "Max Channel" Errors, which is a good sign. I strongly suspect something happened between Beta-1 and beyond that really messed up the channel handling, there's multiple Issues logged that are all related.

scramblr, 2025-03-01:
@wsbush I noticed it says "Build OS          : Linux (amd64 6.2.0-39-generic  [15MB/252MB 6%]" - Are you using the binaries from this release or rebuilding from source?

wsbush, 2025-03-01:
Now I’m confused.  I’m using windows.  Does it say Linux on the beta I was
asked to download and test or the 6.1 release.  I may have downloaded the
incorrect platform.

Thanks, Scott

Randesign, 2025-03-02:
The build OS is irrelevant to the platform you are running on.  All of the platforms can be built from Linux. 
https://github.com/DSheirer/sdrtrunk/wiki/Developing

wsbush, 2025-03-02:
Yes, downgrading to Beta-1 worked.  Tested using this:

https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.1-beta-1

testing results were posted here:  https://github.com/DSheirer/sdrtrunk/issues/2132#issuecomment-2679385769

wsbush, 2025-03-02:
@scramblr ive got a work-around so im not going to roll back the sdrplay api.  Im using 6.1 release, my sdrplay-rspdx2 with my antenna connected, along with an RTL-SDR with no antenna connected.  instead of getting the "Channels exceeded" message, sdrtrunk offloads to the RTL-SDR. even with no antenna, the system works although the log file constantly is saying data is rolling to the RTL-SDR.  when i disconnect my RTL-SDR, only then do i begin getting the "channels exceeded".

Another observation that may or may not help;  with the 6.1 release, i believe the system is treating my sdrplay rspdx as a "duo" model.  regardless which channel i say to use "1" or "2", data is found regardless of which port i use for the antenna.  Ex. ive got my antenna connected to port 2, which works fine.  but i can set the configuration to use port 1, and it also works fine.  i can move the antenna to 1 or 2 and it doesnt make a difference as long as its connected to a port.  here is the interesting part.  ive got an HF only antenna on port 1.  something with SDRPlay must be searching and find  800 MHZ signals then use the found port for traffic.

wsbush, 2025-03-03:
@scramblr i stand corrected, log file shows on startup that it correctly detected my sdrplay.

scramblr, 2025-03-04:
Awesome.

Randesign, 2025-03-06:
Thanks @wsbush for the log files. Couldn't have found this bug without them.

PR #2169 (2025-03-09): Fix adding available channel to wrong queue
Available channel was added to P1 queue instead of P2 queue. Should close #2132.
closed 2025-03-09 by DSheirer