DSheirer/sdrtrunk #2134: P25: Restrict tuner from tuning to calls from specific talkgroups

ZStrikeGit, 2025-01-19:
**Is your feature request related to a problem? Please describe.**

I monitor a P25 site (MPSCS) for my county, and dont really have enough dongles to cover every frequency. I only want to monitor specific talkgroups, 
and it sucks when a dongle that could be used to recieve a call for a talkgroup want to listen to is instead recieving on a different frequency for a talkgroup i don't want to monitor.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
What I would like is the option to essentially restrict the tuner from tuning to calls from specific talkgroups so that it can tune to more "important" channels

I believe this could be quite a simple feature to implement, and would love to see it in the future!

Radiotech012, 2025-01-19:
Can't you do this by adding unwanted talkgroups to the alias list and then locking them out using the 'Listen' switch?

kernworks, 2025-01-20:
I believe @Radiotech012 has the solution you need.

Define the talkgroups you don't want to hear in the alias list you are using for the site. Then toggle listen to off.

Alternatively you could define priorities. The lower the number the more important the talkgroup is. If two different talkgroups start at the same time then the one with the lowest priority will be heard first.

ZStrikeGit, 2025-01-23:
Switching the talkgroups muted will mute the call, but the tuner still tunes to that channel regardless, which is exactly what I don't want

Radiotech012, 2025-01-23:
Ahh, ok..

Well in that case, it sounds like the talkgroup 'listen/do not listen' feature may need a tweak. I can't see why the tuner would need to switch to an active grant channel if that particular talkgroup was muted with the (do not) 'Listen' switch - and if the Streaming and Record switches were also deactivated for that talkgroup.

The issue is that the 'listen' switch only mutes the demodulated audio, it doesn't effect streaming, recording, or apparently grant channel monitoring.

As you point out, maybe the easiest solution is to implement an 'ignore' function in the Alias list, to ignore any voice/data channel grants for particular talkgroups or even radio IDs 🤔

That said, maybe the 'priority' option essentially does the same thing, by switching/monitoring high priority talkgroups and moving low priority groups to the bottom of the list.

Maybe using the (do not) 'Listen' and also making the unwanted talkgroups a lower priority would work?