DSheirer/sdrtrunk #2135: Event filter view saved on exit & single tuner R820T use on trunkig networks that have greater than 2.4Mhz spread

Radiotech012, 2025-01-19:
Hi guys,

Could you add a way to save the Event filter config, so that it will remember the active fields during shutdown/restart?

Also, not sure if this is possible..

My local P25 trunking network operates over about a 4Mhz spread, between 416-420Mhz.
The control channels, typically operate around 419/420 Mhz and the traffic channels can be as low as 416Mhz.
A single R820T dongle cannot cover this spread natively.

However, if I want to monitor only one talkgroup, is there any reason why the R820T centre freq couldnt be shifted down one or two Mhz,
during a channel grant, for a desired talkgroup, then switch back up to the control channel, once the channel grant expires?

I know this wont work if you want to monitor several talkgroups at the same time, but for those that are only interested in monitoring one 
company/agency, this may be a useful feature to keep the hardware cost as low as possible.

kernworks, 2025-01-20:
I can't speak to the first part of your question.

For the second part, that would only really work if the talkgroup sticks on the same frequency every time.

Any system that has a control channel, the control channel has to be monitored.

SDRTrunk will shift the monitored spectrum automatically to cover the channel grants but not at the expense of not being able to monitor the control channel.

Radiotech012, 2025-01-21:
Ahhh yes, I forgot that some channel grants can change during a call. Although, this usually only occurs at the end of a dispatch/unit transmission. Theoretically, if a dispatch/field unit ended a transmission and the channel grant ended, the tuner could switch back to the CC and then obtain & switch to the next grant channel for the response call - maybe... 🤔

I guess it depends on the tuning/settling time of the R820T tuner.

Be interesting to try.