DSheirer/sdrtrunk #2136: Simplified Console GUI / Display

Radiotech012, 2025-01-19:
Hi guys, loving SDRtrunk.

I would love to see a simplified Console/display options that can be implemented on a small 7" HDMI touch display.

For many, I believe, once the local networks/channels and aliases are configured, a simple, large font screen option would be fantastic.

IOW, a display/touch screen GUI that is similar in nature to the old Uniden HomePatrol, for mobile use, that you can easily activate or lockout system or talkgroups easily.

Something like the attached, but in a 7" widescreen format.

[image]

Anyone else thinks this would be useful?

If not, disregard 😁

Cortexian, 2025-01-25:
I would love a UI like this. Could easily deploy sdrtrunk on something like a Raspberry Pi 5 and a little touchscreen and have your own mobile scanner setup pretty easily.

Morpheus636, 2025-04-24:
Exchoing this. I've got SDRTrunk installed on a Pi 5 with the touchscreen and it would be really nice to have a UI that's better suited to that.