DSheirer/sdrtrunk #2141: Streaming will not auto reconnect stays (Connecting) icecast

medic345, 2025-01-22:
**sdrtrunk Version**
6.1 final

**Describe the bug**
Noticing an issue if connection drops out even for a few seconds the streaming status changes to (Connecting) I need to manually go into streaming settings and disable then re enable the stream for it to reconnect again. 

**To Reproduce**
Set up a stream to icecast then disconnect your network connection then reconnect briefly. 

**Expected behavior**
Auto reconnect when network restored

**Desktop (optional - complete the following information):**
 - OS: Ubuntu: 24.10
 - CPU Cores: 8 cores
 - RAM: 64gb

ajcoll5, 2025-08-20:
This appears to be a duplicate of #1892.

ajcoll5, 2025-11-25:
Should be resolved with https://github.com/DSheirer/sdrtrunk/commit/b472c68b807b68ed48d0b30f01a44c9bee9c21a7

closed 2025-11-25 by DSheirer