DSheirer/sdrtrunk #2142: Duplicate call detection not working

Radiotech012, 2025-01-25:
Latest stable & nightly versions:

Im tuned to three different local p25 Phase 1 control channels, on one local Phase 2 network.

Despite having the 'Duplicate call Detection' activated in the Call Management tab, I'm still getting duplicated/echoed audio from
two active voice channels, from two different sites, with the same group talkgroup. On some patched talkgroups, im getting 500ms delayed audio on the two active voice channels and it sounds terrible.

I have identical talkgroups showing in Left & Right audio channels and duplicated (with echo) audio coming out of the speakers.

Is anyone else experiencing this?

Radiotech012, 2025-01-25:
Ok, I think I've tracked the issue down..

On my local P25 trunking network, not all voice calls are time synchronised across different sites.

In other words, a voice/group call will begin on one local site, then a few 10's to a few hundred mS later, the same call will begin on a neighbouring site. It's the same call on the same talkgroup/radio ID, but the demodulated audio is offset by several mS, to several hundred mS.

This seems to be a data routing, or maybe different P25 base station vendor issue, with my my local network??

As the traffic channel grants/calls are offset slightly by time, the 'Duplicate Call Detection' function doesn't seem to be able to pick up on it and then ignore/mute the second site/call.

You essentially get the same talkgroup coming through the left & right speakers, but with a minor, to major audio delay.

I think the solution is to constantly monitor the active talkgroup and ensure that neighbouring sites are ignored/muted (if they subsequently become active on that talkgroup/ID), for the duration of the first call. 

It seems like the 'Duplicate Call Detection' only operates at the moment, if the voice channel grants on different sites, are closely time synchronised.

Radiotech012, 2025-01-25:
Disregard. May have been a local network issue. Will keep monitoring.

closed 2025-01-25 by Radiotech012

DSheirer, 2025-01-25:
The logic that sdrtrunk follows is:

1. Are both calls active at the same time? (but they don't have to be synchronized)
2. Do the calls have the same 'System' value in each of their channel configurations?
3. Do the calls have the same Talkgroup value?

If yes to all 3, then flag one of the calls (the second one to start) as duplicate.

If duplicate call detection by Radio ID is enabled in the preferences, then step 4 is to compare the Radio ID values. 

This logic continues for subsequent (3rd, 4th, etc) duplicate call detection and flagging.