DSheirer/sdrtrunk #2147: DMR Decoder does not function properly

Tbolt10033, 2025-02-07:
**sdrtrunk Version**
master-branch

**Describe the bug**
When using the DMR decoder on a conventional DMR system, the decoder fails to work properly showing CRC Idle errors and has issues when a call is made on the system where it is erratic.

**To Reproduce**
Steps to reproduce the behavior:
Use the DMR decoder on a DMR conventional system.

**Expected behavior**
For the DMR decoder to correctly decode the DMR system data and not have errors.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Randesign, 2025-02-07:
As of this date, the master-branch is not decoding P-25 either. The master-branch is the same as nightly builds and is not guaranteed to work. 

If you need a working release, try one listed in the right side pane.

DSheirer, 2025-02-09:
@Tbolt10033 this looks like a RAS protected system and it looks like it's decoding it correctly.   RAS uses a non-standard CRC checksum and therefore sdrtrunk can't calculate the checksum.

Was there something more that you think that  sdrtrunk should be doing?

Tbolt10033, 2025-02-09:
> [@Tbolt10033](https://github.com/Tbolt10033) this looks like a RAS protected system and it looks like it's decoding it correctly. RAS uses a non-standard CRC checksum and therefore sdrtrunk can't calculate the checksum.
> 
> Was there something more that you think that sdrtrunk should be doing?

It was working fine before the latest nightly, on 0.6.1 it works fine but after that version, it does not work correctly.

closed 2025-04-23 by Tbolt10033