DSheirer/sdrtrunk #2155: Version 6.1 and sdrplay RSPdx R2 unable to handle channels efficiently causing failure.

wsbush, 2025-02-19:
Running version 6.1

Phase 2 

using SDRPlay RSPdx R2 ( and also an RTL-SDR for diagnostics )

defined 6 traffic channels

preferred tuner is RSPdxR2

ISSUE:  when running 6.1 with only the RSPdxR2, it runs for about a minute.   The program is responsive but no output.   i have logs attached.   

I added an rtl-sdr dongle, restarted sdrtrunk, Both SDR's show as available with the sdrplay as priority.   the rtl-sdr does NOT have antenna connected.  with both connected, SDRTrunk 6.1 operates without a visible issue.

im including logs with event logs for both conditions

----------------------------------------------------------
Attachments:

[sdrplay only console.log](https://github.com/user-attachments/files/18876195/sdrplay.only.console.log)
[with sdrplay and rtlsdr console.log](https://github.com/user-attachments/files/18876196/with.sdrplay.and.rtlsdr.console.log)

[only sdrplay event logs.zip](https://github.com/user-attachments/files/18876197/only.sdrplay.event.logs.zip)
[with sdrplay and rtlsdr console event logs .zip](https://github.com/user-attachments/files/18876198/with.sdrplay.and.rtlsdr.console.event.logs.zip)

scramblr, 2025-02-24:
Use https://github.com/DSheirer/sdrtrunk/tree/v0.6.1-beta-1 this version and let me know if it works for u.

wsbush, 2025-02-24:
im testing with the beta.  this version cant load the SDRPlay API.  went back and verified 6.1 release still does.  ive attached log.

20250224 123846.065 [main] WARN  i.g.d.s.t.sdrplay.api.Version - Unrecognized SDRplay API version [3.15]  [10MB/28MB 38%]
20250224 123846.065 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library is not available.  [10MB/28MB 38%]
2

[sdrtrunk_app.log](https://github.com/user-attachments/files/18948407/sdrtrunk_app.log)

scramblr, 2025-03-01:
This should be condensed into one issue

DSheirer, 2025-03-09:
SDRPlay API version 3.15 support was added in versions 0.6.1-beta2 and is supported in later releases.  Version 0.6.1-beta1 does not support API version 3.15, so your last comment is fully expected ... you're trying to use the wrong version of sdrtrunk for the SDRPlay API version that is installed on your computer.

You might also be having a problem with limited computer resources.  You're running Windows 7 on a 4-core machine that's only offering 3GB of RAM allocation to sdrtrunk.

With that in mind, what is the issue that you're reporting here?

wsbush, 2025-03-12:
release 6.1 doesnt work using only an SDRPlay device.  if i add another SDR like an rtl-sdr, system works and logs show lots of channels redirected to the rtl-sdr.   

resolution to issue 2132 may also have corrected this?  Let me know if there is a build to test and ill download.

DSheirer, 2026-11-07:
closing.  sdrtrunk now uses jSDRPlay library.  Please advise if the issue is not resolved.

closed 2026-11-07 by DSheirer