DSheirer/sdrtrunk #2157: Occasional talkgroup locks onto an audio channel

blainek-ew, 2025-02-21:
**sdrtrunk Version**
2025-02-15 Nightly

**Describe the bug**
Occasional issue where a talk group will fail to release and block one of the audio channels ( left or right). Stopping the channel does not clear the condition.  The program must be restarted.

**To Reproduce**
have been unable to reproduce at will but comes up at least once a day.  Channel is a P25 Phase 1 Simulcast LSM. using left and right channels.

**Expected behavior**
audio call should close out properly so the audio channel can be used for the next call.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
Will try to capture this going forward.

**Desktop (optional - complete the following information):**
 - OS: Windows 10
 - CPU Cores: 8 cores (16 threads)
 - RAM:32GB

**Additional context**
Add any other context about the problem here.

blainek-ew, 2025-02-21:
Issue happened again just now.  Left audio channel locked by ACPD APCOM1.  No related active calls.  All audio processing coming through right channel.   Several logs attached along with a screen shot for what it's worth.

[sdrtrunk_app.log](https://github.com/user-attachments/files/18914125/sdrtrunk_app.log)

[20250221_140150_sdrtrunk_thread_dump.log](https://github.com/user-attachments/files/18914111/20250221_140150_sdrtrunk_thread_dump.log)

[20250221_140144_sdrtrunk_processing_diagnostic_report.log](https://github.com/user-attachments/files/18914113/20250221_140144_sdrtrunk_processing_diagnostic_report.log)

[image]

blainek-ew, 2025-02-21:
Stopped the active channel and showing left channel still locked on non existent stream.

[image]

iLoveGin, 2025-02-21:
Hello, I am still trying to get screen shots but I am having the same issue. The Control channel goes to idle while it switched to an audio channel, once the audio transmission is done the control channel is restored. I am still trying to replicate it. It happened to me twice in a row after rebooting yesterday but it has not happened since. Take care!

blainek-ew, 2025-02-21:
Yea its about once per day for me.  Our P25 system is pretty active.   When
it happens just generate the debug reports and attach them all.   Multiple
data points should be helpful in tracking it down.

blainek-ew, 2025-02-27:
This happened again.  This time I stopped all active channels and created logs

[image]

[sdrtrunk_app.log](https://github.com/user-attachments/files/19002819/sdrtrunk_app.log)
[20250227_002531_sdrtrunk_thread_dump.log](https://github.com/user-attachments/files/19002820/20250227_002531_sdrtrunk_thread_dump.log)
[20250227_002525_sdrtrunk_processing_diagnostic_report.log](https://github.com/user-attachments/files/19002818/20250227_002525_sdrtrunk_processing_diagnostic_report.log)

DSheirer, 2025-03-09:
Left audio thread stacktrace:

```
Thread[#70,sdrtrunk audio output LEFT thread 1,5,main] TIMED_WAITING
	at java.base@23.0.1/java.lang.Object.wait0(Native Method)
	at java.base@23.0.1/java.lang.Object.wait(Unknown Source)
	at java.desktop@23.0.1/com.sun.media.sound.DirectAudioDevice$DirectDL.write(Unknown Source)
	at app//io.github.dsheirer.audio.playback.AudioOutput.playAudio(AudioOutput.java:409)
	at app//io.github.dsheirer.audio.playback.AudioOutput.processAudio(AudioOutput.java:532)
	at app//io.github.dsheirer.audio.playback.AudioOutput$AudioSegmentProcessor.run(AudioOutput.java:784)
	at java.base@23.0.1/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base@23.0.1/java.lang.Thread.runWith(Unknown Source)
	at java.base@23.0.1/java.lang.Thread.run(Unknown Source)
```

blainek-ew, 2025-03-09:
What would be causing TIMED_WAITING for a channel that has been stopped?  Seems like something did not close properly.

PR #2191 (2025-03-13): #2157 Audio Output Gets Stuck & Loss of Gain/Mute Control
Closes #2157 

Resolves audio output getting 'stuck' when the thread is blocking on an attempt to write audio to the data line and that blocking call is not executing.  

Also, updates previous attempt at fixing a stalled data line by closing and reopening the line, without having to recreate the data line, which was causing related issues with the gain and muting behavior.
closed 2025-03-13 by DSheirer